revtel/react-native-nfc-manager #105: NfcManager.onStateChange callback is executed only if NfcManager.start is called before

filippoitaliano, 2018-10-24:
Is this the expected behaviour? I would like to use `NfcManager.onStateChange` when the NFC is caught disabled (with `NfcManager.isEnabled`) during app initialization. In such case I prompt the user to enable it and I try to re-start the service immediatly after `NfcManager.onStateChange` catch the enabling event. Is this possible without manually call and handle the rejection of `NfcManager.start` before the `NfcManager.onStateChange` call?

Thanks in advance!

whitedogg13, 2018-10-24:
Yes, it's the expected behavior, we do want the `NfcManager.start` is called before any other methods in this library. It is a safe operation even if the NFC is disabled.

filippoitaliano, 2018-10-24:
Ok, got it.

closed 2018-10-24 by filippoitaliano