revtel/react-native-nfc-manager #106: Add Support for Ultralight

sahaya4us, 2018-10-24:
I have used [react-native-nfc-manager](https://github.com/whitedogg13/react-native-nfc-manager) in my react native project to read NFC tags.

But I am not able to read Ultralight tags.  It seems, It is not supported yet.  Could you please add Ultralight tag support, as most of the devices in my country uses [MifareUltralight](https://www.mifare.net/en/products/chip-card-ics/mifare-ultralight/) . If this library support [MifareUltralight](https://www.mifare.net/en/products/chip-card-ics/mifare-ultralight/) it will be more helpful to the NFC  community.

subrajits, 2018-10-24:
+1 for Ultralight support request

sahaya4us, 2018-10-26:
Hi @whitedogg13 , Thank you so much. It is working perfectly for [mifare-ultralight-support](https://github.com/whitedogg13/react-native-nfc-manager/tree/mifare-ultralight-support).

whitedogg13, 2018-10-26:
Thanks for feedback! We will merge it into master and release soon.

whitedogg13, 2018-10-30:
Hi, we have published v1.1.0 to support `MifareUltralight` officially, thanks for contributing!

closed 2018-10-30 by whitedogg13