revtel/react-native-nfc-manager #109: iPhone XR

GlobeTracker, 2018-11-01:
Does this library support iPhone XR?  I am receiving the error 'Not support in this device' when calling start.  Thanks.

whitedogg13, 2018-11-01:
Hi, thanks for reporting, it sounds like a bug, we will fix it soon!

whitedogg13, 2018-11-04:
Not able to find a iPhone XR / XS to test against. Can some one help on this one?

whitedogg13, 2018-11-05:
Hi @GlobeTracker , I just test this on XS, iOS 12.1.1, and it's working correctly. Can you please check your actual iOS version? (BTW, this library won't work in simulator)

muhammadarsal, 2018-11-05:
Hi! I tested the package on iPhone 7 iOS 12.0.1, getting the same error.

whitedogg13, 2018-11-06:
Hi @muhammadarsal thanks for feedback! 

Currently I'm not able to find a iPhone 7 to test. However, from the implementation side, the `isSupport` is simply doing this:
```objective-c
if (@available(iOS 11.0, *)) {
    if (NFCNDEFReaderSession.readingAvailable) {
        result = YES;
    }
}
```
Do you have a change to take a look?

Thanks!

GlobeTracker, 2018-11-08:
Hi @whitedogg13, thanks so much for looking into this.  I am using a physical device with 12.0.1; not a simulator.  I will check back in the morning with further testing results.

GlobeTracker, 2018-11-08:
Hi @whitedogg13, I have pasted the stack and code below.
Stack:
"react": "16.0.0-alpha.12",
"react-native": "^0.50.4",
"react": "16.0.0-alpha.12",
"redux-saga": "^0.15.6",
"redux-thunk": "^2.2.0",
I can include the full package.json if it is helpful.

Code:
  startNFC = () =>{
    NfcManager.start({
      onSessionClosedIOS: () => {
        console.log('ios session closed')
      }
    })
      .then(result => {
        console.log('start OK', result)
      })
      .catch(error => {
        console.warn('device does not support nfc!')
        this.setState({supported: false})
      })
  }
// END OF CODE

After calling start, I get the error I reported above.

GlobeTracker, 2018-11-09:
Hi @whitedogg13, my provisioning profile was incorrect.  I am now getting the supported result!  Thank you.

whitedogg13, 2018-11-09:
It's awesome! Good for you!

closed 2018-11-09 by whitedogg13

Sotir-Akkroo, 2019-07-02:
Hey! I got the same error @GlobeTracker for Xr on a physical device. I know the issue is closed, but I assume few more people can end up here.

What was your problem/solution to this?

AlexS64, 2021-04-27:
Same problem as @sotir-ls.
Tested on an IPhone XR, IOS-Version: 14.4

When calling the start() Method it returns an error "Not supported in this device"