revtel/react-native-nfc-manager #110: fatal error: 'React/RCTEventEmitter.h' file not found

suraj-shanbhag, 2018-11-07:
Hi , i'am getting this error when I try to build in ios ,Android works fine
[image]
I have followed this order:
- npm i --save react-native-nfc-manager 
- react-native link react-native-nfc-manager
- setup Capabilities,Plist
- Pod install
My React version is "16.4.1" and RN version:"0.55.4"

whitedogg13, 2018-11-07:
Hi, thanks for reporting this issue. Can you try it again without using Pod?

christianchown, 2018-11-14:
@suraj-shanbhag have you recently upgraded React Native? I often get these kind of errors when I do, and they are all false negatives.

Try the following from the root of your project to clear all your directories and caches and reinstall your libraries
```
rm -rf node_modules
rm -rf ~/.rncache
rm -rf ~/Library/Developer/Xcode/DerivedData
rm -rf ~/Library/Caches/CocoaPods
rm -rf ios/CocoaPods
yarn cache clean     # or npm cache clean --force
yarn                 # or npm install
cd ios
pod install
```

If that doesn't work, go into Xcode and go Project->Scheme->Edit scheme of your project, go to "Build" and untick Parallelize Build. (You can reenable it once you're building again)

whitedogg13, 2019-06-07:
close due to no further activities

closed 2019-06-07 by whitedogg13