revtel/react-native-nfc-manager #113: How to check when a "Beam" has been transmitted?

rogercoll, 2018-11-14:
I am trying to make and app and I am using Android Beam between two stations. The "requestAndroidBeam" works great but I can not get when the transmission of the bytes has finished. Here is the function:
`_requestAndroidBeam = (info) => {
        let {isWriting} = this.state;
        if (isWriting) {
            return;
        }
        let bytes;
        bytes = buildTextPayload(info);
        this.setState({isWriting: true});
        NfcManager.setNdefPushMessage(bytes)
            .then(() => {console.log("Beaming...")})
			.catch(err => console.warn(err))
    }`

Inside the "then" you only know that is beaming, not if it has finished.

whitedogg13, 2018-11-14:
@rogercoll Thanks for reporting.

It's possible through native Android API `setOnNdefPushCompleteCallback`, but we haven't added it into this library yet. I will try to implement this when I'm available and keep you posted.

whitedogg13, 2021-02-02:
as Android Beam has been deprecated, we will remove Android NFC p2p related feature

closed 2021-02-02 by whitedogg13