revtel/react-native-nfc-manager #116: ISO / IEC 1443 support?

GlobeTracker, 2018-11-16:
Hi @whitedogg13,  I would like to begin by thanking you for this library.  It works wonderfully!  I am able to obtain an ISO / IEC 1443-3 tag using this library with an Android device.  However, I am not able to capture the same Tag using the library on iOS.  NFCCore is said to include ISO / IEC 1443-2 and -3 support.  

To further isolate my use case, I downloaded and used the NXP-TagInfo App and received the same result (works on Android but not iOS 12.1).  Is there something special that I have to do, or is a problem with the Tag manufacturer?  I have included a screenshot of the NXP TagInfo screenshot running on Android.

Thanks in advance.

[image]

poison, 2018-11-17:
Hi @GlobeTracker,

As far as I know, NFC support on iOS is very limited. At the time of writing, I only know of Ndef support. Too bad that iOS is extremely slow in adapting common technologies. Let me know if you find other working examples of different NFC technology on iOS, then we can look into it.

Best regards,
Nicolas

GlobeTracker, 2018-11-17:
Thanks @whitedogg13,  I was able to test your library on iOS with this NFC product.  Good news for Globe Tracker, because we intend on using these smart tags for some of our clients. Perhaps this could help you look into it!

http://label.averydennison.com/content/dam/averydennison/lpm-responsive/europe/english/documents/product-overviews/product/pharmaceutical/functionality/prov-tt-sensor-plus-2-en.pdf

GlobeTracker, 2018-12-05:
Hi @whitedogg13, Any luck with the tag I provided?  If you PM me with an address, I can send you a working Avery Dennison sample tag

whitedogg13, 2018-12-09:
Hi @GlobeTracker sorry for the late response, kinda busy recently.

From the NXP tag reader screenshot, the tag is an ISO14443AA / MifareUltralight, and it should be supported by iOS, as reported from some SO discussion:
* https://stackoverflow.com/questions/44572827/detect-iso-iec-14443-mifare-ultralight-or-classic-nfc-cards-with-corenfc

Beside this, the iOS core nfc only deals with NDEF, so you should double check if your tag contains valid NDEF messages. 

Sadly, no further advice can be provided beside above points, since NFC support for iOS is quite limited.

Thanks!

whitedogg13, 2019-06-07:
close due to no further activities

closed 2019-06-07 by whitedogg13