revtel/react-native-nfc-manager #121: "New tag collected" when beaming data

rogercoll, 2018-12-11:
Is it possible to prevent the phone to open the "New tag collected" screen? I think that this screen is provided by Samsung. I just searched if there is any way to disable it but anyone works for me.

whitedogg13, 2018-12-11:
Hi, @rogercoll , you're right about the 'new tag collected' view, it's an default Activity provided by Android when no other Activities intercept the NFC related intents. 

To prevent the default view, you can simply call `registerTagEvent` (please refers to `example/App.js` in this repo) to signal that your app would like to handle the NFC intent, thus the default view wouldn't popup.

rogercoll, 2018-12-11:
@whitedogg13 Hey, yes that was what I thought but I does not work. This is my code:
```
_onTagDiscovered = tag => {
        console.log('Tag Discovered', tag);
		let text = this._parseText(tag);
		let aux = JSON.parse(text);
		if(aux.name == this.state.whoamI){
			this._isok(aux);
			this.setState({texte: text});
		}
    }
	
	_startDetection = () => {
        NfcManager.registerTagEvent(this._onTagDiscovered)
            .then(result => {
				console.log('Read OK', result)
            })
            .catch(error => {
                console.warn('registerTagEvent fail', error)
            })
    }
```

tereshchenkoartyom94, 2018-12-13:
Just call a NfcManager.registerTagEvent() to prevent Android to discover a tag. 
Try to await NfcManager.registerTagEvent() and that is all

whitedogg13, 2019-06-07:
close due to no further activities

closed 2019-06-07 by whitedogg13