revtel/react-native-nfc-manager #123: Types for ndef-lib

natethebosch, 2018-12-14:
Idk if you want this, but I've created typings for ndef-lib. If you want it - feel free to use it, if not - no harm done. Nice work on this project, it's worked well for me.

```
declare module 'react-native-nfc-manager/ndef-lib/index.js' {
  type NdefRecord = any;
  type ISOLangCode = "en" | string;
  type IgnoredParameter = any;
  type URI = string;

  export const text: {
    encodePayload: (text: string, lang?: ISOLangCode, encoding?: IgnoredParameter) => NdefRecord;
    decodePayload: (data: NdefRecord) => string;
  };

  export const uri: {
    encodePayload: (uri: URI) => NdefRecord;
    decodePayload: (data: NdefRecord) => string;
  };

  export const util: {
    stringToBytes: (string: any) => any[];
    bytesToString: (bytes: any) => string;
    bytesToHexString: (bytes: any) => string;
    toHex: (i: any) => any;
    toPrintable: (i: any) => string;
  };

  export function stringify(data: number[], separator: string);
  export function encodeMessage(records: NdefRecord[]);
  export function decodeMessage(bytes: any[] | Buffer): NdefRecord[];
  export function textRecord(text: string, lang?: ISOLangCode, encoding?: IgnoredParameter): NdefRecord;
  export function uriRecord(uri: URI, id: any): NdefRecord;
}
```

natethebosch, 2018-12-14:
Updated to fit into existing types declaration

whitedogg13, 2018-12-14:
Hi @natethebosch it's awesome! Can you do a PR? I will be very happy to merge it.

whitedogg13, 2018-12-14:
Just see the PR, thanks a lot!

closed 2018-12-14 by whitedogg13

nickwebha, 2019-01-24:
Unless I am misunderstanding this has been merged and pushed (source: https://www.npmjs.com/package/react-native-nfc-manager/v/1.1.4).

I am still getting this:
```
source/library/nfc.ts:58:14 - error TS2339: Property 'isType' does not exist on type '{ text: { encodePayload: (text: string, lang?: string | undefined, encoding?: any) => NdefRecord; decodePayload: (data: any[]) => string; }; uri: { encodePayload: (uri: string) => NdefRecord; decodePayload: (data: any[]) => string; }; ... 5 more ...; uriRecord(uri: string, id?: any): NdefRecord; }'.

58  		if ( Ndef.isType( tag, Ndef.TNF_WELL_KNOWN, Ndef.RTD_TEXT ) ) {
    		          ~~~~~~

source/library/nfc.ts:58:32 - error TS2339: Property 'TNF_WELL_KNOWN' does not exist on type '{ text: { encodePayload: (text: string, lang?: string | undefined, encoding?: any) => NdefRecord; decodePayload: (data: any[]) => string; }; uri: { encodePayload: (uri: string) => NdefRecord; decodePayload: (data: any[]) => string; }; ... 5 more ...; uriRecord(uri: string, id?: any): NdefRecord; }'.

58  		if ( Ndef.isType( tag, Ndef.TNF_WELL_KNOWN, Ndef.RTD_TEXT ) ) {
    		                            ~~~~~~~~~~~~~~

source/library/nfc.ts:58:53 - error TS2339: Property 'RTD_TEXT' does not exist on type '{ text: { encodePayload: (text: string, lang?: string | undefined, encoding?: any) => NdefRecord; decodePayload: (data: any[]) => string; }; uri: { encodePayload: (uri: string) => NdefRecord; decodePayload: (data: any[]) => string; }; ... 5 more ...; uriRecord(uri: string, id?: any): NdefRecord; }'.

58  		if ( Ndef.isType( tag, Ndef.TNF_WELL_KNOWN, Ndef.RTD_TEXT ) ) {
    		                                                 ~~~~~~~~
```

As a stopgap I added this to the bottom of _index.d.ts_:
```
type NdefRecord = any;
type ISOLangCode = "en" | string;
type IgnoredParameter = any;
type URI = string;

export const Ndef: {
  text: {
    encodePayload: (text: string, lang?: ISOLangCode, encoding?: IgnoredParameter) => NdefRecord;
    decodePayload: (data: NdefRecord) => string;
  };

  uri: {
    encodePayload: (uri: URI) => NdefRecord;
    decodePayload: (data: NdefRecord) => string;
  };

  util: {
    stringToBytes: (string: any) => any[];
    bytesToString: (bytes: any) => string;
    bytesToHexString: (bytes: any) => string;
    toHex: (i: any) => any;
    toPrintable: (i: any) => string;
  };

  TNF_WELL_KNOWN: number;
  RTD_TEXT: number;

  isType(record: NdefRecord, tnf: number, type: number);
  stringify(data: number[], separator: string);
  encodeMessage(records: NdefRecord[]);
  decodeMessage(bytes: any[] | Buffer): NdefRecord[];
  textRecord(text: string, lang?: ISOLangCode, encoding?: IgnoredParameter): NdefRecord;
  uriRecord(uri: URI, id: any): NdefRecord;
}
```

This is not, by any means, meant to be extensive but only cover my current needs. Also I am not wild about using _any_ virtually anywhere within this file.

PR #124 (2018-12-14): Add types for ndef-lib
idk if you want this, but here's some types I did up for Ndef-lib. 
closes #123