revtel/react-native-nfc-manager #126: How To Make Sure that requestNdefWrite has been Written to NFC Successfully?

VincentAlexander, 2019-01-07:
Hi,

I got an issue that why my apps gave a success write notification but it's not really completely written yet? Below is the code for my requestNdefWrite function. Why is .then(() => console.log("write success")) called but it's not completely success? For example I remove the NFC card too early, the .then() called, but when I read the NFC card, the data is still the old data, not the new one.

Now to make sure the requestNdefWrite have been successfully written to NFC card, I need to read the NFC card using nfcOnTagDiscovered to make sure the data in NFC card is the correct data.

> this.setState({isWriting: true});
        NfcManager.requestNdefWrite(bytes)
            .then(() => this.setState({suksesWrite: true}))
            .catch(err => {
                this.setState({
                    suksesWrite: false
                })
                console.warn(err)
                this._hideModal()
                this._NfcStopDetection()
                toastError("Proses NFC gagal, silahkan coba lagi!")
            })
            .then(() => this.setState({isWriting: false}))

Can someone give me the solution for this issue? Thank you.

Regards,

Vincent Alexander

VincentAlexander, 2019-01-11:
Hi @whitedogg13 , thank you for your reply. I'm looking forward for the result ya.

whitedogg13, 2019-01-13:
Hi @VincentAlexander I just publish 1.1.4, please check if the issue fixed. Thanks!

VincentAlexander, 2019-01-14:
Hi @whitedogg13 I tried the v1.1.4 and it works perfectly! I tried to remove the NFC card early and it gives error notification now. Great! Thank you for always helping and replying to my issue so far.

closed 2019-01-14 by VincentAlexander