revtel/react-native-nfc-manager #127: Any sample of "transceive"

hosseinpro, 2019-01-09:
Thanks for this useful library.
I'm trying to communicate with a ISO 14443 card on an Android device, but I could not find any example of "transceive" API. This is my code that works but one time I just get event in onTagDiscovered function and another time in getTag function. I mean the line of transceive is not executed every time.

`_startNfc = () => {
    NfcManager.start()
      .then(() => NfcManager.isEnabled())
      .then(enabled => this.setState({ enabled }))
      .then(() => NfcManager.registerTagEvent(this.onTagDiscovered))
        .catch(err => {
        console.warn(err);
        this.setState({ enabled: false });
      });
  };

onTagDiscovered = tag => {
    let i = "hello";
    console.log(i);
    NfcManager.requestTechnology(NfcTech.IsoDep)
      .then(() => NfcManager.getTag())
      .then(() => {
        console.log(JSON.stringify(tag));
        let b = [0x00, 0xa4, 0x00, 0x00];
        NfcManager.transceive(b).then(response => {
          console.log(response);
        });
        NfcManager.closeTechnology();
      })
      .catch(err => {
        console.warn(err);
      });
  };
`

ChusFSB, 2019-03-05:
This code didn´t work for me. It returns "transceive fail".

My code:

 _startNfc = () => {
        NfcManager.start()
            .then (() => NfcManager.registerTagEvent (this.onTagDiscovered,"ret",{ isReaderModeEnabled: true,readerModeFlags:  NfcAdapter.FLAG_READER_NFC_A }))
            .then (() =>NfcManager.requestTechnology(NfcTech.NfcA))
            .catch(err => {
                console.warn(err)
            })
    }

    onTagDiscovered = tag => {
        console.log(tag)
        let b =  [0x00, 0xA6, 0x00, 0x00, 0x00]
        return NfcManager.transceive(b) 
        .then(result => {
            console.log(result)
        }) .catch(err => {
            console.warn(err)
            NfcManager.closeTechnology()
        })
    }

The value of tag is : {"techTypes":["android.nfc.tech.IsoDep","android.nfc.tech.NfcA"],"id":"0123456789"}

Any help please?

MttNajafi, 2019-04-10:
Hi @whitedogg13  I have the same problem.

MttNajafi, 2019-04-12:
@hosseinpro Thank you for your prompt reply, I will try it.

whitedogg13, 2019-06-07:
close due to no further activities

closed 2019-06-07 by whitedogg13