revtel/react-native-nfc-manager #129: NfcManager.isSupported got 1 arguments, expected 2

fabioloreggian, 2019-01-20:
When I call 
```
NfcManager.isSupported('').then((supported) => {
      this.setState({ supported });
      if (supported) {
        this.startNfc();
      }
    });
```

I get this error:
[image]

fabioloreggian, 2019-01-20:
```
"dependencies": {
    "jsencrypt": "^3.0.0-rc.1",
    "lodash": "^4.17.10",
    "moment": "^2.22.2",
    "prop-types": "^15.6.2",
    "react": "^16.4.2",
    "react-native": "0.56.0",
    "react-native-camera": "^1.9.0",
    "react-native-card-flip": "^1.0.4",
    "react-native-code-push": "^5.4.2",
    "react-native-crypto-js": "^1.0.0",
    "react-native-elements": "^0.19.1",
    "react-native-linear-gradient": "^2.4.0",
    "react-native-nfc-manager": "^1.1.4",
    "react-native-phone-call": "^1.0.9",
    "react-native-qrcode": "^0.2.7",
    "react-native-splash-screen": "^3.1.1",
    "react-native-svg": "^6.5.2",
    "react-native-vector-icons": "^6.1.0",
    "react-navigation": "^2.11.2",
    "uuid": "^3.3.2"
  },```

whitedogg13, 2019-01-22:
@fabioloreggian Looks like the native complains about the missing arguments, however, the lib will pass a default one (see `isSuppoorted` method in `NfcManager.js`) so it shouldn't happen. Can you double check whether the installed library is corrupted?

fabioloreggian, 2019-01-22:
@whitedogg13 
That was exactly my thinking, I went through your code and there wasn't any possible way that this error should have been thrown. So I ended up starting a new app and relinked all my native packages and copied over the source code and everything is working fine.

closed 2019-01-22 by fabioloreggian