revtel/react-native-nfc-manager #130: Buffer size issue

hosseinpro, 2019-01-25:
I use this module to transmit APDU with ISO 14443 smart card on Android. Every this works but when I want to read a data from the card which is 227 byte I get "transceive fail" error.

Is there any max buffer size limit? Do you support ISO extended length?

hosseinpro, 2019-01-25:
I thinks I found the problem. According to this: [https://developer.android.com/reference/android/nfc/tech/IsoDep.html#isExtendedLengthApduSupported()](url)

Some (or most) Android phone does not support extended length like Nexus S (my phone).

whitedogg13, 2019-01-29:
@hosseinpro thanks for reporting this! I will bridge the `isExtendedLengthApduSupported` API into next release.

whitedogg13, 2019-02-26:
Hi @hosseinpro , we have added an API `getMaxTransceiveLength` in latest release (and the IsoDep should be supported), so that we will be able to detect the max buffer size, please give it a try!

closed 2019-03-27 by whitedogg13