revtel/react-native-nfc-manager #132: how can I authenticate with MifareUltralight ?

bluetiger30, 2019-02-04:
Hi , is there an authentication function to authenticate with MifareUltralight C? or is the transceive function the authentication function for MifareUltralight, if so can you kindly help me out with an example?

whitedogg13, 2019-02-14:
Hi @bluetiger30 , as far as I know, Android doesn't expose such API.

Android API for `MifareUltralight` is listed here: https://developer.android.com/reference/android/nfc/tech/MifareUltralight, and no authentication related ones are included.

JamesZhange, 2019-05-29:
I can use NFC tools app like 'NFC Tools Pro' to authenticate my MifareUltralight C, how they do that?

whitedogg13, 2019-06-07:
@JamesZhange sorry for the late response.

It's possible to use the low-level `transceive` API to approach this feature, but it requires a deep understanding for `MifareUltralight C` tag command set. 

Since I'm not an expert for this tag, I will label this issue as help wanted and see if someone can help us!

Thanks!

whitedogg13, 2019-10-24:
close inactive issue

closed 2019-10-24 by whitedogg13