revtel/react-native-nfc-manager #134: How to read NTAG 213 locked NDEF

mkharibalaji, 2019-02-11:
Hi @whitedogg13 ,
            How to read a NTAG 213 which is password protected or locked ?. Right now I am not able to properly read it.The Programmed URL comes with extra bytes before and after the URL.

whitedogg13, 2019-02-11:
Hi, @mkharibalaji Please refer to this [SO link](https://stackoverflow.com/questions/36395457/read-write-to-nfc-tag-with-password-protection). For tag with memory protection, you usually need to perform authentication manually, normally it's done through the low level `transceive` API.

mkharibalaji, 2019-02-11:
Hi @whitedogg13,
         Thanks a lot buddy.

          If you shed some light on how i could  use react-native-nfc-manager to read and write on a locked NTAG 213 it would be great.

whitedogg13, 2019-02-14:
Sorry, we mainly focus on android / iOS related API bridging, and cannot help on the individual tag platform. (BTW, I think `makeReadOnly` won't help).

I think you will need to read the tag data sheet, like this one: https://www.nxp.com/docs/en/data-sheet/NTAG213_215_216.pdf, then you can figure out how to use low-level `transceive` to communicate with the tag.

mkharibalaji, 2019-02-14:
Thanks a lot @whitedogg13.

Will definitely give it a try.

closed 2019-02-26 by whitedogg13