revtel/react-native-nfc-manager #136: Refusing to install package with name "react-native-nfc-manager" under a package npm ERR! also called "react-native-nfc-manager". Did you name your project the same npm ERR! as the dependency you're installing?

MssMaster, 2019-02-13:
Getting above error while installing 'react-native-nfc-manager' inside example project.Please let me know what i am doing wrong.Also if i import android project in android studio and update it's gradle versions it does not show any configuration to run the code.Please let me know how i can fix it ?

MssMaster, 2019-02-13:
Able to install the plugin by changing name of application to nfc-manager in package-json.js class. but still it does not display any run configuration in android studio.

MssMaster, 2019-02-14:
Any One there ?

whitedogg13, 2019-02-14:
@MssMaster I'm not sure I understand your issue. You should first create your own RN app, and npm install / react-native link this library. The example directory contains no full project.

MssMaster, 2019-02-14:
Thanks for your reply,I am new to this, Now i am able to run the code and can see the Ui of screen but still start detection is not doing anything, i have nfc card with me i am taping it behind the device but the it is not getting detected by application.Even i tried to put alert inside _startDetection() method it's not getting called.

MssMaster, 2019-02-14:
code of app.js class

MssMaster, 2019-02-15:
yes, I am doing in same way.

whitedogg13, 2019-02-16:
From the code you listed, the result will be passed into `_onTagDiscovered` callback~

MssMaster, 2019-02-18:
yes, But I am not getting any value for the tag, seems it never goes into onTagDiscovered method.

MssMaster, 2019-02-19:
can you please look into the above code and let me know if there is anything missing or wrong?

whitedogg13, 2019-10-24:
close inactive issue

closed 2019-10-24 by whitedogg13