revtel/react-native-nfc-manager #137: setTimeout function for transceive

hosseinpro, 2019-02-13:
Hi 
I found that react-native-nfc-manager does not export Android NFC setTimeout function. I want to do a long processing task on card and react-native-nfc-manager returns "transceive fail".
When I remove long processing task from card, it works. Also I test my card with a USB reader with my laptop and there is no error.
According to this:
https://developer.android.com/reference/android/nfc/tech/NfcA.html#setTimeout(int)
And this:
https://stackoverflow.com/questions/6877650/changing-the-nfca-timeout-length-in-android
the default timeout is 300ms and we can extend it with setTimeout function.

whitedogg13, 2019-02-14:
@hosseinpro thanks for the suggestion, we will add this API soon!

whitedogg13, 2019-02-26:
Hi @hosseinpro we have added support for `setTimeout` in the latest 1.2.1, please have it a try!

closed 2019-03-27 by whitedogg13

MatteoAlberghini, 2020-12-18:
Hi, is this feature only for 1.x version? I don't see the `setTimout` function for 2.x

arafat-al-mahmud, 2021-01-16:
Couldn't find setTimeout in 2.2.1

suyog0391, 2021-02-16:
Missing this functionality in 2.2.1 latest version

GitKokoszka, 2021-03-24:
Yes, it is missing for me too.

whitedogg13, 2021-03-24:
Hi all, sorry for the late response, you guys are right, this one is indeed missing in type definition.

I have added it back in `2.3.1` and `3.1.1`, thanks!

Ellathet, 2023-03-15:
How can use the timeout method? 
I try to use with await and try, how it is done in the doc:

```
  async function readNdef() {
    try {
      await NfcManager.requestTechnology(NfcTech.Ndef);
      const tag = await NfcManager.getTag();
      await NfcManager.setTimeout(60000 /* 1 minute in ms */);
    } catch (ex) {
      console.error('error')
    } finally {
      NfcManager.cancelTechnologyRequest();
    }
  }
  ```