revtel/react-native-nfc-manager #138: React Native NFC Manager v 1.2.0 iOS Build Failure

mkharibalaji, 2019-02-19:
Hi @whitedogg13 ,
          The new version with the below spec is failing in ios build,

/Users/mkharibalaji/reactnative/fermynt/node_modules/react-native-nfc-manager/ios/NfcManager.m:132:84: Interface type 'NSDictionary' cannot be passed by value; did you forget * in 'NSDictionary'?

"react": "16.6.3",
    "react-native": "0.58.4",
    "react-native-camera-kit": "^7.3.3",
    "react-native-elements": "^1.0.0",
    "react-native-navigation": "^2.12.0",
    "react-native-nfc-manager": "^1.2.0",
    "react-native-vector-icons": "^6.3.0",
    "react-redux": "^6.0.0",
    "redux": "^4.0.1",
    "redux-thunk": "^2.3.0"

whitedogg13, 2019-02-19:
@mkharibalaji Thanks for reporting! I will fix it ASAP.

whitedogg13, 2019-02-19:
@mkharibalaji The fixed version 1.2.1 has been published, really appreciated for your early reporting!

closed 2019-02-19 by whitedogg13