revtel/react-native-nfc-manager #142: Callback in NfcManager.registerTagEvent(Callback) doesn't trigger

pareshprajapati1990, 2019-02-26:
In NfcManager.registerTagEvent(this.onTagDiscovered), onTagDiscovered never gets called.

mantcz, 2019-03-08:
Check again if you have the library installed properly. I have just installed it and the callback works on both iOS and Android.

pareshprajapati1990, 2019-03-11:
> Check again if you have the library installed properly. I have just installed it and the callback works on both iOS and Android.

Hi,
Thanks for answering to my query.

Can you please tell me which two devices did you use to test whether the callback is working properly or not?

mantcz, 2019-03-11:
Hey @pareshprajapati1990 

I have used iPhone X with iOS 12 and Android device, Samsung 8 I believe.

whitedogg13, 2019-06-07:
Hi all, since it's the basic functionality for this library, I believe it should work if the configuration is correct. I will close this issue for now, please re-open or fire another one if the issue remains!

closed 2019-06-07 by whitedogg13