revtel/react-native-nfc-manager #143: format is not working

panmissl, 2019-03-02:
Hi, I am using your library and found an issue: format is not working.
I got this error:
[image]

"fail to apply ndef formatable tech."

I have tried several apps but only "Nfc tools" can format nfc tag.

could you give some advise or fix this issue?

Thanks

whitedogg13, 2019-03-04:
Hi @panmissl the error message means that the tag you try to format doesn't have a `NdefFormatable` tech in it. 

You should inspect the `tag` object obtained by `registerTagEvent` to see all available NFC technologies, and only use certain NFC tech when it's appliable.

panmissl, 2019-03-04:
> Hi @panmissl the error message means that the tag you try to format doesn't have a `NdefFormatable` tech in it.
> 
> You should inspect the `tag` object obtained by `registerTagEvent` to see all available NFC technologies, and only use certain NFC tech when it's appliable.

Hi whitedogg, I am still confused about this: If the error message means that the tag doesn't have a NdefFormatable tech in it, then how can the "Nfc tools" format it successfully?

Also, when I use "Nfc tools" erase it, the tag shows it support "NfcA" and "MifareUltralight" techs, now any tools and you library CAN"T read or write the tag;  And then I use "Nfc tools" format it, the tag shows it support "NfcA", "MifareUltralight", and "Ndef" techs, so I can use your library to read and write it for now.

whitedogg13, 2019-06-07:
Hi @panmissl , sorry for pending so long.

My guess is that the `Nfc tools` app implements the NDEF formatting in lower level: they probably send raw commands directly to the tag, rather than depends on [this](https://developer.android.com/reference/android/nfc/tech/NdefFormatable.html#format(android.nfc.NdefMessage)) Android API.

In our library, we depends on Android to do the formatting for us, and it can only do this when `NdefFormatable` tech is included in the tag event. 

Of cause, this library can send raw commands to do formatting as well (via the `NfcManager.transceive`), but this requires us to understand the raw NFC commands and protocols. For most app developers like us, it won't be the case, but for an app like `Nfc tools`, this should be their know-how.

Hope that makes sense!

whitedogg13, 2019-10-24:
close inactive issue

closed 2019-10-24 by whitedogg13

paichato, 2022-02-04:
Update to the latest version @3.13.0 and try again.

> Hi whitedogg, I am still confused about this: If the error message means that the tag doesn't have a NdefFormatable tech in it, then how can the "Nfc tools" format it successfully?
>
> Also, when I use [...]