revtel/react-native-nfc-manager #148: Native module cannot be null on import of NfcManager in RN 0.59.2

robseaman, 2019-03-27:
import NfcManager, { Ndef } from 'react-native-nfc-manager';
is resulting in a 'Native module cannot be null.' error on ios using react-native-nfc-manager 1.2.1.

I have verified that it's linked and entitlements are correct. I tried unlinking removing, adding an linking again.

I was previously successfully using RN 0.57.1 and 0.6.0. In that version I had libNfcManager.a in `Link Binary with Libraries` but it doesn't exist on the new version. Along with that there isn't an NfcManager.xcodeproj installed in the Libraries in Xcode on the new version.

whitedogg13, 2019-03-27:
Hi @robseaman , I just do some quick test regarding this, however it's working for me.

My steps are:
```
react-native init NfcTest
cd NfcTest
npm install --save react-native-nfc-manager
react-native link react-native-nfc-manager
```
Both the xcodeproj and the .a are placed as they're expected. 

If the issue persists, maybe you can perform some manual link for now?

robseaman, 2019-03-27:
Thanks @whitedogg13 for getting on this so quickly. It does create everything correctly when I start a new project from scratch and install it first. It's not hard for me to recreate my project so that will be good enough. I don't have any idea how to recreate the situation I was in so I'll close this.

closed 2019-03-27 by robseaman