revtel/react-native-nfc-manager #152: getLaunchTagEvent() gets called every reload.

Tatogv, 2019-04-03:
Hi! I love your library but found this issue.

Whenever I reload the app or even sometimes when the app is in the background and I open it again, the getLaunchTagEvent method returns a tag.
This only happens when the app is launched via a tag.

Thank you!

whitedogg13, 2019-04-12:
@Tatogv Thanks for reporting this issue and also sorry for the late response. I will check it out and get back to you.

whitedogg13, 2019-05-11:
Hi, @Tatogv really sorry for such a late response!

I think the problem you reported might be an expected behavior. Because the `getLaunchTagEvent` will call Android's `getIntent` under the hood, and `getIntent` always return the intent starts the targeting activity, so it will remain the same.

Maybe you can try to limit the `getLaunchTagEvent` call, to only during app launching.

Thanks!

whitedogg13, 2019-06-07:
close due to no further activities, please re-open or fire a new one if the issue remains!

closed 2019-06-07 by whitedogg13