revtel/react-native-nfc-manager #153: Android: Getting error 'You can only issue one request at a time' while calling requestNdefWrite method

parvezq, 2019-04-18:
Below given is react-native code for writing to the Mifare Ultralight NFC card:

`initiateWrite = async () => {
   const supported = await NfcManager.isSupported(NfcTech.MifareUltralight)
   if (supported) {
     console.warn('UltraLight Card is Supported')
     NfcManager.registerTagEvent((tag) => {
       console.warn(tag)
       NfcManager.requestTechnology(NfcTech.MifareUltralight).then(() => {

       })
       this.write('123456789');
     })

   }

 }
write =(text) => {
   var data = []        
         for (let i = 0; i < text.length && i < NfcManager.MIFARE_BLOCK_SIZE; i++) {
           data[i] = parseInt(text.charCodeAt(i));
         }
         console.warn(data)
var output = {}
         var i,j,temparray,chunk = 4, k = 0;
         for (i=0,j=data.length; i<j; i+=chunk) {
           temparray = data.slice(i,i+chunk);
           console.warn(temparray)
            NfcManager.requestNdefWrite(temparray, {format: true})
           .then(() => {
             console.warn('write done. ')
           })
           .catch((error) => console.warn(error))
           break
           k += 1
         }
 }`

parvezq, 2019-04-19:
> Below given is react-native code for writing to the Mifare Ultralight NFC card:
>
> `initiateWrite = async () => {
> const supported = await NfcManager.isSupported(NfcTech.MifareUltralight)
> if [...]

This issue has been resolved as we found the problem in our code itself.

closed 2019-04-19 by parvezq

jungRoit, 2021-03-22:
@parvezq can you share the fix that you used for this issue. I'm facing a similar issue

visheshsprinzo, 2021-04-08:
@parvezq  yes same with me!