revtel/react-native-nfc-manager #155: Change the value after first form.

QuantomFlame, 2019-05-09:
@whitedogg13 Hello, this library is very cool. But i have one issue about it. 

I have three screens (First - Is form when you can create/add worker, Second - Is form when the worker can take tool, Third- Is form when the worker can return tool).

When I loaded the first form and put the card I receive successfully message and the card is work. When I loaded the second or the third form I and put the card I don't receive nothing and the scanner not read the card. And for example when I load thte second form I go back and go forward the card starts working. I use react-navigation. Is it bug?

whitedogg13, 2019-05-11:
Hi @QuantomFlame , sorry for the late response.

I don't think I understand your issue. Do you mean that the NFC tag detection fails across multiple screens? Per my understanding, since `react-navigation` is a JS side navigation library, these two libraries should play well.

whitedogg13, 2019-10-24:
close inactive issue

closed 2019-10-24 by whitedogg13