revtel/react-native-nfc-manager #157: Behavior of NfcManager.getLaunchTagEvent doesn't get worked, If I scan the tag, it does not launch the app.

KinjalDroid, 2019-05-16:
Behavior of NfcManager.getLaunchTagEvent doesn't get worked, If I scan the tag, it does not launch the app.

So I have to open the app manually, And even if NfcManager.getLaunchTagEvent is called after opening app, It always retrun null tag.

SO please correct me.. What should be changed in code to launch the app when I scan the Tag. Should I change in AndroidManifest.xml file ? or when NfcManager.getLaunchTagEvent() method should be called ?

whitedogg13, 2019-06-07:
@KinjalDroid sorry for the late response.

It depends on the data you put inside your NDEF, but for general cases, you should do both. Because those two steps are for different purposes:
* `AndroidManifest.xml` can wake your app up directly (it's Android's native behavior, not RN related)
* Then, once your RN app launched, call `getLanuchTagEvent` to retrieve the parsed launching tag (it's actually an Intent, in Android's term)

For example, if you want your app to be launched when a NDEF contains an URI like `http://myapp.com`, you should add following intent into your `AndroidManifest`:
```XML
<intent-filter>
    <action android:name="android.nfc.action.NDEF_DISCOVERED"/>
    <category android:name="android.intent.category.DEFAULT"/>
    <data android:scheme="http"
              android:host="myapp.com" />
</intent-filter>
```
The, call `getLaunchTagEvent` when your app starts up. (the resolved tag might be null if your app is not launched via tag)

I think you're asking a great question, and I'm sorry that we didn't explain this well in the doc. Please kindly tell me if above suggestion works!

whitedogg13, 2020-02-26:
close inactive issue

closed 2020-02-26 by whitedogg13