revtel/react-native-nfc-manager #158: Is it possible to have a custom mimeType for an NDEF_DISCOVERED?

miltonTouchdown, 2019-05-16:
i want to have a custom mimeType for read a ndef, like to: 
mimeType="application/vnd.hello.sample.nfcmanager".

The project already have text/plain and uri. 

I need this for avoid conflicts with another applications on the devices.

Many thanks in advance for your answers

whitedogg13, 2019-06-07:
Hi @miltonTouchdown sorry for the late response.

You can add following intent filter into your `AndroidManifest.xml` to achieve this:
```XML
<intent-filter>
    <action android:name="android.nfc.action.NDEF_DISCOVERED" />
    <category android:name="android.intent.category.DEFAULT" />
    <data android:mimeType="application/vnd.hello.sample.nfcmanager" />
</intent-filter>
```

Thanks!

whitedogg13, 2020-02-26:
close inactive issue

closed 2020-02-26 by whitedogg13