revtel/react-native-nfc-manager #160: Android Error on NfcManager.start()

MaxBolotnyi, 2019-05-27:
Just installed the lib and tested with the following code to see if it works.

I am using redux-saga, here is the sample code that did not work for me.

`function* nfcWatcher(api) {
  while (true) {
    try {
      const action = yield take(TRANSFERS_SCAN_NFC);

      // will fall back to the catch here if the NFC is not supported on device.
      const nfcProcess = yield call(NfcManager.start);

      console.log(nfcProcess)
      console.log('nfcProcess')

    } catch {
      Alert.alert(
        'NFC Warning',
        'This device does not support NFC',
        [
          {
            text: 'OK',
            onPress: () => NavActions.pop(),
          }
        ]
      )
      console.log('CATCH - nfcWatcher!!!!!!!!');
    }
  }
}`

The following error happens (screen attached)

[image]

If I comment this out in your library, the app does not crash anymore.
[image]

Please, advice.

whitedogg13, 2019-06-07:
Hi @MaxBolotnyi sorry for the late response. 

I'm not using saga, but from the code I can guess what's going wrong.
In your code, you have extract the `NfcManager.start` function like this: 
```javascript
const nfcProcess = yield call(NfcManager.start);
```
You can change it to:
```javascript
const nfcProcess = yield call(() => NfcManager.start());
```

The reason is, In your original code, the `this` is not bond to `NfcManager` (because we didn't use arrow function for NfcManager.js, we will consider to change that in later release).

Please retry and tell me if there's still anything going wrong.

whitedogg13, 2020-02-26:
close inactive issue

closed 2020-02-26 by whitedogg13