revtel/react-native-nfc-manager #162: Calling requestTechnology results in crash

flowridr, 2019-06-04:
**Using the following:**

NfcManager.start()
            .then(() => NfcManager.isEnabled())
            .then(enabled => this.setState({enabled}))
            .catch(err => {
                console.warn(err);
                this.setState({enabled: false})
            })

NfcManager.registerTagEvent(tag => {})

**Results in:**

Crash:
community.revteltech.nfc.TagTechnologyRequest.connect
TagTechnologyRequest.java, line 54
java.lang.NullPointerException: Attempt to invoke virtual method 'boolean java.lang.String.equals(java.lang.Object)' on a null object reference

Tag type used NfcA

whitedogg13, 2019-06-07:
Hi @flowridr , sorry for the late resopnse. There's no calling for `NfcManager.requestTechnology` in above code, maybe you're posting the wrong code?

whitedogg13, 2020-02-26:
close inactive issue

closed 2020-02-26 by whitedogg13