revtel/react-native-nfc-manager #164: Reading ePassport

mquintal, 2019-06-15:
Hi,

I'm trying to implement an app which needs to read data from ePassport.
I'm able to get the tag but then Im not sure how I read the information from the it.
I know that I need to authenticate (using password number, birth date and expiry date) and then read its data.
If you somebody can give some feedback/example how I can do it using this lib would be great.

Thank you!

whitedogg13, 2019-06-16:
Hi @mquintal I'm not familiar with ePassport format. However, I found some links might be useful to your app:
* https://stackoverflow.com/questions/21849601/android-nfc-read-data-from-epassport
    * Explain how to auth against the ePassport tag in brief, and some references 
* https://github.com/tananaev/passport-reader
    * Sample Android app to read ePassport (in Java)

Thanks!

ross-ewing, 2019-10-25:
@mquintal mquintal did you have any luck passing authentication to the passport - I know we need the id from the number data of birth etc but Im seeing anywhere in the read functionality to pass in an encryption id?  All the provided links are android related and working on ios.

closed 2021-05-26 by github-actions[bot]