revtel/react-native-nfc-manager #165: NFC Writing on IOS 13

balaah, 2019-06-15:
Apple recently announced that they will support NFC writing on IOS 13. I'm working on an app and I would love to enable this feature for my IOS customers, however my IOS skills are rather rudimentary.

Are you planning on integrating NFC writes in this library in the future? If so, can you estimate how hard this will be / how long this would take?

whitedogg13, 2019-06-16:
@balaah Yes, we will definitely integrate these features into this library, but we don't have a concrete schedule for now. 

Since there're lots of new stuffs in iOS 13, if you can tell us which feature you're interesting most, we can consider implementing them first.

Thanks!

balaah, 2019-06-16:
@whitedogg13 I'm still trying to wrap my head around the new features. I think iOS 13 offers an equivalent of Android's transceive command. That would be my most desired feature as I would like to  read and write from specific locations in the tag's memory. A transceive-like command would allow me to target those locations. I think Apple calls this writing "ADPUs" but I'm not certain.

Other than that, the most important feature for me would be the ability to write NDEF records.

Your library is amazing! :)

ChristopheBelpaire, 2019-06-16:
Indeed, same for me. Having a transceive command would be super great!

Thanks for the library :)

jafri, 2019-07-23:
Writing NDEF records is most interesting!

balaah, 2019-07-29:
@whitedogg13 Do you have a rough timeline for basic support of iOS 13? I still believe the transceive command makes most sense as a starting point as all higher level functions can be bootstrapped once it is possible to send transceive commands (e.g., writing NDEF records).

whitedogg13, 2019-08-02:
@balaah about the schedule, we hope to align the official iOS 13 release date (by convention, it's most likely to be at the end of September). And just like you, we believe lots of app can benefit from these new low-level NFC APIs. 

However, because of the differences of the two native platforms, some APIs you previously find in Android might not be available in iOS. For example, in iOS there's no generic `transceive` for NFC A/B/V/F, instead Apple chooses to directly support Mifare / Felica with the ability to send custom command.

whitedogg13, 2019-08-22:
Hi, I have some updates to share regarding the upcoming iOS 13!

I have just published a new beta release `2.0.0-beta.2`, which includes some of the most wanted iOS 13 functionalities:
* NDEF writing
* get tag UID
* sendMifareCommand

The codes are in the `v2` branch.

No documentations yet, but I have added two examples in the example dir:
* AppIOS13 (for uid and mifare)
* AppIOS13Ndef (for NDEF writing)

Though the API is still unstable and might change in anytime, but I still hope to share with you. Any suggestions and PRs are welcome!

Thanks!

jberns88, 2019-08-22:
Is there any plans for ISO15693 support? I've been waiting many moons to be able to read/write NfcV tags on iOS :)

whitedogg13, 2019-08-24:
@jberns88 yes, we will definitely has iso15693!

jberns88, 2019-08-24:
Brilliant thanks for your hard work :-)

whitedogg13, 2019-09-10:
Hi all, thank you for using react-native-nfc-manager! Some updates for you.

I have just published `2.0.0-beta.3`. In this new version, I finally consider the API to be stable. 

It's much better than v1 in my own opinion. Some important changes:
* no longer need to call `registerTagEvent` before `requestTechnology`!
* can pass multiple techs into `requestTechnology`, and it will resolve to the detected tech
* provide `setEventListener` and export `NfcEvents` to listen to specific native event
* `setAlertMessageIOS` to update the iOS alertMessage

For example, requet NDEF tech + write NDEF will look like this:
```javascript
    let resp = await NfcManager.requestTechnology(NfcTech.Ndef);
    console.warn(resp); // should be 'Ndef'

    let ndef = await NfcManager.getNdefMessage();
    console.warn(ndef);

    let bytes = buildUrlPayload('https://www.revteltech.com');
    await NfcManager.writeNdefMessage(bytes);
    console.warn('successfully write ndef');
```

Or we can do some custom mifare command to read the first page:
```javascript
    let tech = Platform.OS === 'ios' ? NfcTech.MifareIOS : NfcTech.NfcA;
    let resp = await NfcManager.requestTechnology(tech);
    console.warn(resp);

    // the NFC uid can be found in tag.id
    let tag = await NfcManager.getTag();
    console.warn(tag);

    if (Platform.OS === 'ios') {
      resp = await NfcManager.sendMifareCommandIOS([0x30, 0x00]);
    } else {
      resp = await NfcManager.transceive([0x30, 0x00]);
    }
    console.warn(resp);
```

Full example can be found in `example` directory, I also briefly explain the difference between v1 and v2 in `APIv2.md`.

I will continue to work on rest technologies available in iOS13. Look forward for you feedback or PR!

Justin2997, 2019-09-10:
Hi @whitedogg13 Very good job !

I am confuse have as for the use case of AppV2Mifare.js vs AppV2.js. Is maby me that don't realy know about NFC tag. If there is a big difference we should mentionne it into APIv2.md.

Also, I experiment some problem with the exemple at APPV2.js where my Iphone XR (IOS 13) get underfined for tag.id, but my Android phone (Android 9) get the correct uid. Did you experience the same problem ?

Let me know if I can help for someting, I have some free time.

Thanks a lot !

whitedogg13, 2019-09-11:
@Justin2997 Regarding the `tag.id` problem, it's becaues iOS only give us the `id` field when using specific NFC tag technology, such as [mifare](https://developer.apple.com/documentation/corenfc/nfcmifaretag/3183038-identifier?language=objc). However for Android platform, the `id` field is always available when any tag found.

The difference between example `AppV2` and `AppV2Mifare`, is that the former has been considered as a "NDEF reading only" use case, and the second one demonstrates "using specific NFC technology" (I use the term "basic" and "advanced" in APIv2.md to explain that concept)

What's specific NFC technology? For example, if we're writing an app to read from our custom tag (such as a custom loyalty program card, access control card), we ofcause know their tag technology (ex: Mifare), so we can send custom command to it for our specific usage (ex: open door lock, add loyalty credit...), that's what `AppV2Mifare` example want to show. 

Thanks!

jafri, 2019-09-11:
@whitedogg13 

You can use writeLock on iOS to makeReadOnly() work on iOS too: https://developer.apple.com/documentation/corenfc/nfcndeftag/3075573-writelock

sallar, 2019-10-18:
@whitedogg13 thanks for the great library. Im using V2 Beta 4, and it works great on Android with an NfcV (`iso15693`) tag. But I'm trying to read the same tag on iOS with no luck :( Is it possible to read such tags?
Thanks!

jberns88, 2019-10-18:
@sallar he's not completed that yet but is working on it. I'm also waiting patiently :-)

jafri, 2020-01-07:
@whitedogg13 can we see writeLock implemented on iOS please :)

whitedogg13, 2020-01-08:
Hi all, I just published `2.1.0-beta.0`, which contains most ISO15693 support for iOS 13. 

Most of the core NFC APIs are implemented except for the multiple blocks related ones.

For example, please see [here](https://github.com/whitedogg13/react-native-nfc-manager/blob/ios-15693/example/AppV2Iso15693.js).

Since I only test a small part of the API for this beta release, please feel free to contribute or fire issues, and the codes are in branch `ios-15693`.

Thanks!

sallar, 2020-01-18:
Thanks @whitedogg13 this is amazing. Anything I can do to help with the multi block reading?

whitedogg13, 2020-01-18:
@sallar Definitely! Please feel free to implement the multi-block APIs for 15693, I will be very happy to accept the PR.

Jagaa9a, 2020-03-24:
Hi @whitedogg13 

I am working on a react-native application project that needs to read data from a type B NFC card. Its data transfer protocol is ISO/IEC14443-4:2008TypeB and it needs to pass Initialization and anticollision step first. I am very new to NFC and I am reading your document and googling all-day, can't find documents related to my NFC card type. Is it even possible to read data from those NFC type B cards? Or should I try it with both native codes?

Thanks,

smartareas, 2020-06-15:
@whitedogg13 it's possible to write in the NFC label the AFI value with iOS?
Thank you very much!

impl-kano, 2020-09-17:
> Hi @whitedogg13
> 
> I am working on a react-native application project that needs to read data from a type B NFC card. Is it even possible to read data from those NFC type B cards? Or should I try it with both native codes?

Hi, Have you found any solution regarding Type-B? Thanks.

closed 2021-02-02 by whitedogg13