revtel/react-native-nfc-manager #167: Clear NfcManager.getLaunchTagEvent

marcolangebeeke, 2019-06-17:
Hi,

The getLaunchTagEvent works great for me, both when the app is not running and when in the background.. or even in the foreground it takes the intent with the nfc tag, **job wel done**! (it looks like it starts a new instance of the app though, is that even possible??)

But that's not my question. When my app is launched with an NFC scan, and I navigate to the functionality that handles the NFC tag, I want to clear the tag afterwards. **So, how do I clear the intent that contains the NFC tag?** In a native Android Activity I guess it would be cleared once you finish the activity and navigate somewhere else in your app, but in React you only have one main activity. Is there a function in NfcManager for this? I could not find it in the readme and source-code.

Thanks!

simonasdev, 2020-02-18:
also searching for answer to this

simonasdev, 2020-02-18:
duplicate of #152. Seems like you would need to manually clear native android's intent data/action, after it has been consumed. I might make a PR that allows this

marcolangebeeke, 2020-02-19:
Thanks @simonasdev !

simonasdev, 2020-02-19:
I've changed my mind :D After some research found that RN in version 0.62 already handles `NDEF_DISCOVERED` actions that contain uris and dispatches `url` events to Linking module, so the implementation of `NfcManager.getLaunchTagEvent` becomes obsolete.
If you want to get these `url` events by simpy using `Linking.addEventListener('url', () => {})` in earlier version of RN you can add this code in your `MainActivity.java`:
```java
//...
import android.content.Intent;
import android.nfc.NfcAdapter;
import android.net.Uri;
import com.facebook.react.modules.core.DeviceEventManagerModule;
import com.facebook.react.bridge.ReactContext;

public class MainActivity extends ReactActivity {
    //...
    @Override
    public void onNewIntent(Intent intent) {
        String action = intent.getAction();
        Uri uri = intent.getData();

        if (uri != null && (Intent.ACTION_VIEW.equals(action) || NfcAdapter.ACTION_NDEF_DISCOVERED.equals(action))) {
            ReactContext context = getReactInstanceManager().getCurrentReactContext();

            DeviceEventManagerModule module = context.getNativeModule(DeviceEventManagerModule.class);
            module.emitNewIntentReceived(uri);
        }

        super.onNewIntent(intent);
    }
}
```
(I'm using react-native-navigation so my implementation was a bit different)

closed 2021-05-26 by github-actions[bot]

whitedogg13, 2021-07-10:
Hi all, regarding this issue, we have a much better solution in the latest `v3.9.0`. 

In this release, we can retrieve the background tag via `getBackgroundTag` (can be used for the initial tag event as well), and we can also listen to the following tags via `DiscoverBackgroundTag` event. And the `backgroundTag` can be cleared via `clearBackgroundTag` API (though I don't spot a valid use case for this one).

These APIs are available for both `iOS` and `Android`.

The whole idea is easier to understand via following code snippet:

```javascript
function onBackgroundTag(bgTag) {
  // bgTag on Android is a complete tag with info such as id, techs
  // bgTag on iOS contains only a `ndefMessage` property
  if (bgTag) {
    navigation.navigate('TagDetail', {tag: bgTag});
  }
}

// get the initial launching tag
const bgTag = await NfcManager.getBackgroundTag();
onBackgroundTag(bgTag);

// listen to other background tags after the app launched
NfcManager.setEventListener(
  NfcEvents.DiscoverBackgroundTag,
  onBackgroundTag,
);
```

A full example app can be found in [NFC ReWriter](https://github.com/revtel/react-native-nfc-rewriter).

BTW, We hope you can migrate to `v3`, and please feel free to fire any questions regarding your migration.