revtel/react-native-nfc-manager #170: not work on react native version 0.60.0

gooking, 2019-07-06:
not work on react native version 0.60.0

version 0.60.0 is up on ,We hope you can support it.

whitedogg13, 2019-07-06:
@gooking Thanks for reporting, it should be related to the breaking change introduced by RN 0.60. I will check it out and fix it soon.

fadyThePharaoh, 2019-07-09:
Can someone mention the latest stable version of react-native that is  compatible with react-native-nfc-manager ??
Thanks in advance.

whitedogg13, 2019-07-09:
hi @fadyThePharaoh , it should work for all versions below RN 0.60 due to the breaking changes. I personally run several apps sing RN 0.56.0 with this lib and it works well.

whitedogg13, 2019-07-13:
Hi @gooking @fadyThePharaoh , I have just updated a beta version targeting RN 60, currently versioned as beta to avoid affecting major RN 5x users. Please give it a try via:

`npm install react-native-nfc-manager@beta`

Related setup info is also updted in readme file, thanks!

closed 2019-07-13 by whitedogg13

svvitale, 2019-07-16:
Tested on 0.60.3 and it works great!  Thanks, @whitedogg13 .