revtel/react-native-nfc-manager #173: read creditcard

deividnn, 2019-07-23:
its possible read credit card whith iso7816emv?

whitedogg13, 2019-08-22:
Yes, it should work with any iso7816 compliant NFC cards.

stultusstudio, 2019-10-31:
@MaxWilliamJF @deividnn Did you end up getting credit card working with this library?

toxipat, 2019-11-29:
is there any way to get the credit card number?

toxipat, 2019-12-24:
> Yes, it should work with any iso7816 compliant NFC cards.

How?

Bad-Listener, 2020-09-25:
Any updates on this one?

I am a little lost...

I succesfully scan the tag of the card, but I cannot understand how to get i.e. cardnumber

selcukitmis, 2020-11-30:
@whitedogg13 how can I read credit card ? 
I tried the example project but it's didn't work.
Can you give an example.

gabrielmoncea, 2021-02-23:
I am also interested in this feature :)

whitedogg13, 2021-02-23:
@gabrielmoncea it's awesome, please feel free to submit a PR! Since most of this feature relates to APDU passing using IsoDep NfcTech, maybe we can consider start this from an example.

gabrielmoncea, 2021-02-24:
I would love to help. Can I do this only through JavaScript?

whitedogg13, 2021-02-24:
@gabrielmoncea sure, I believe it's possible!

deanpretorius, 2021-04-12:
@whitedogg13 I would also love to help with this feature! A tutorial seems to be in demand, and I would love to implement this feature in an app I am designing.

closed 2021-06-28 by github-actions[bot]

ronaldocala, 2022-02-15:
Has there been any solution to this ?

MaxWilliamJF, 2022-02-22:
> @MaxWilliamJF @deividnn Did you end up getting credit card reading working with this library?

No =(

ws7one, 2022-09-24:
This seems to be a commonly requested implementation of this library. Has there been any solution to this?

lmatamoros, 2023-01-12:
Visa example 

"node-emv": "^1.0.22"
"react-native-nfc-manager": "^3.14.2"

```
import NfcManager, { NfcTech } from 'react-native-nfc-manager'
import emv from 'node-emv'

readVisaCreditCard = async () => {
	try {
		NfcManager.cancelTechnologyRequest()
    } catch (error) { }
	
	try {
		const commands = [
			"00A404000E325041592E5359532E444446303100",
			"00A4040007A00000000310100E",
			"80A800002383212800000000000000000000000000000002500000000000097820052600E8DA935200"
		]
		
		await NfcManager.requestTechnology([NfcTech.IsoDep])
		
		const responses = []
      
		for (let i = 0; i < commands.length; i++) {
			const resp = await NfcManager.isoDepHandler.transceive(this.toByteArray(commands[i]))
			responses.push(resp)
		}

		if (responses && responses.length > 2) {
			const r = await this.getEmvInfo(this.toHexString(responses[2]))
			if (r) {
				const cardInfo = this.getCardInfoVisa(r)
				if (cardInfo) {
					return {
						card: cardInfo.card,
						exp: cardInfo.exp
					}
				} else {
					return null
				}
			} else {
				return null
			}
		} else {
			return null
		}
    } catch (error) {
		return null
    } finally {
		NfcManager.cancelTechnologyRequest()
	}
}

getEmvInfo = (info) => {
    return new Promise((resolve) => {
		emv.describe(info, (data) => {
			if (data) {
				resolve(data)
			} else {
				resolve(null)
			}
		})
    })
}

toByteArray = (text) => {
	return text.match(/.{1,2}/g).map(b => {
		return parseInt(b, 16)
	})
}

toHexString = (byteArr) => {
    return byteArr.reduce((acc, byte) => {
		return (
			acc + ('00' + byte.toString(16).toUpperCase()).slice(-2)
		)
    }, '')
}

getCardInfoVisa = (responses) => {
    let res
    let end = false
    for (let i = 0; i < responses.length; i++) {
		const r = responses[i]
		if (r.tag === "77" && r.value && r.value.length > 0) {
			for (let j = 0; j < r.value.length; j++) {
				const e = r.value[j]
				if (e.tag === "57" && e.value) {
					const parts = e.value.split("D")
					if (parts.length > 1) {
						res = {
							card: parts[0],
							exp: parts[1].substring(0, 4)
						}
						end = true
					}
				}

				if (end) {
					break
				}
			}

			if (end) {
				break
			}
		}
    }
    return res
}
```

lmatamoros, 2023-04-11:
Mastercard example

`
import NfcManager, { NfcTech } from 'react-native-nfc-manager'
import emv from 'node-emv'

readMasterCardCreditCard = async () => {
	try {
		NfcManager.cancelTechnologyRequest()
    } catch (error) { }
	
	try {
		const commands = [
			"00A4040007A00000000410100E",
			"80A8000002830000",
			"00B2011400",
			"00B2010C00",
			"00B2012400",
			"00B2022400"
		]
		
		await NfcManager.requestTechnology([NfcTech.IsoDep])
		
		const responses = []
      
		for (let i = 0; i < commands.length; i++) {
			const resp = await NfcManager.isoDepHandler.transceive(this.toByteArray(commands[i]))
			responses.push(resp)
		}

		if (responses && responses.length > 3) {
			const r = await this.getEmvInfo(this.toHexString(responses[3]))
			if (r) {
				const cardInfo = this.getCardInfoMasterCard(r)
				if (cardInfo) {
					return {
						card: cardInfo.card,
						exp: cardInfo.exp
					}
				} else {
					return null
				}
			} else {
				return null
			}
		} else {
			return null
		}
    } catch (error) {
		return null
    } finally {
		NfcManager.cancelTechnologyRequest()
	}
}

getEmvInfo = (info) => {
    return new Promise((resolve) => {
		emv.describe(info, (data) => {
			if (data) {
				resolve(data)
			} else {
				resolve(null)
			}
		})
    })
}

toByteArray = (text) => {
	return text.match(/.{1,2}/g).map(b => {
		return parseInt(b, 16)
	})
}

toHexString = (byteArr) => {
    return byteArr.reduce((acc, byte) => {
		return (
			acc + ('00' + byte.toString(16).toUpperCase()).slice(-2)
		)
    }, '')
}

getCardInfoMasterCard = (responses) => {
    let res
    let end = false
    for (let i = 0; i < responses.length; i++) {
		const r = responses[i]
		if (r.tag === "70" && r.value && r.value.length > 0) {
			for (let j = 0; j < r.value.length; j++) {
				const e = r.value[j]
				if (e.tag === "5A" && e.value) {
					if (!res) {
						res = {
							card: e.value
						}
					} else {
						res.card = e.value
					}

					if (res.card && res.exp) {
						end = true
					}
				}

				if (e.tag === "5F24" && e.value) {
					if (!res) {
						res = {
							exp: e.value
						}
					} else {
						res.exp = e.value
					}

					if (res.card && res.exp) {
						end = true
					}
				}

				if (end) {
					break
				}
			}

			if (end) {
				break
			}
		}
    }
    return res
}
`

tamilmani12, 2023-04-18:
@lmatamoros 
can you help me to get  debit card  details for both visa and master I am getting responses like this for debit card 
lenHex: 85 lenBin: 10000001 ---> len is more than 1 byte
 LOG  length (decimals): 0 offset: 14 - this is for the substring of emv_data
 LOG  [{"length": "85", "tag": "69", "value": []}] lllll
 LOG  [{"length": "85", "tag": "69", "value": []}] r
 LOG  [{"length": "85", "tag": "69", "value": []}]

tamilmani12, 2023-04-22:
@lmatamoros please share some ideas
thank you

Ali12275, 2023-05-11:
@lmatamoros  can you help me to get debit card  details for both visa and master i am getting response like this for debit card
lenHex: 82 lenBin: 10000010 ---> len is more than 1 byte
LOG length (decimals): 0 offset: 14 - this is for the substring of emv_data
LOG [{"length": "82", "tag": "6A", "value": []}] r

Noitidart, 2023-07-06:
This is very helpful, thanks for sharing. Just redirecting users from this issue to here - https://github.com/revtel/react-native-nfc-manager/issues/632

Is the techniques here similar to what Stripe does? Stripe makes it sound like Android and iOS can scany any (not just visa or mastercard) - https://stripe.com/en-au/terminal/tap-to-pay (screenshot below)

[image]

tamilmani12, 2023-09-21:
> @lmatamoros can you help me to get debit card details for both visa and master i am getting response like this for debit card lenHex: 82 lenBin: 10000010 ---> len is more than 1 byte LOG length [...]

did you resolved this?

Laulhus, 2023-11-23:
> @lmatamoros can you help me to get debit card details for both visa and master i am getting response like this for debit card lenHex: 82 lenBin: 10000010 ---> len is more than 1 byte LOG length [...]

I managed to read my VISA debit card following @lmatamoros implementation but updating the commands varialble to the following:

`const commands = [
        '00A404000E325041592E5359532E444446303100',
        '00A4040007A000000003101000',
        '80A80000238321F620C00000000000000100000000000007240000000000097823112300194E172C00',
        '80A800002383212800000000000000000000000000000002500000000000097820052600E8DA935200',
        '80CA9F1700',
        '80CA9F3600',
      ];`

reggiegutter, 2023-12-07:
Hi, @lmatamoros and @Laulhus, where did guys find the commands necessary for each card type? I've been searching online and can't find anything. It would be super helpful if you could share a bit more.

Also, is it possible to verify the PIN using this library? I'm still trying to understand all that's involved with accepting contactless payment with an Android Phone. From what I understood until now, if I can verify the PIN than this is enough to build something like a SoftPOS to receive payments.

lmatamoros, 2023-12-07:
> Hi, @lmatamoros and @Laulhus, where did guys find the commands necessary for each card type? I've been searching online and can't find anything. It would be super helpful if you could share a bit more.
> 
> Also, is it possible to verify the PIN using this library? I'm still trying to understand all that's involved with accepting contactless payment with an Android Phone. From what I understood until now, if I can verify the PIN then this is enough to build something like a SoftPOS to receive payments.

You can't get the CVV or the cardholder's name using NFC.

Laulhus, 2023-12-07:
> Hi, @lmatamoros and @Laulhus, where did guys find the commands necessary for each card type? I've been searching online and can't find anything. It would be super helpful if you could share a bit [...]

Hi @reggiegutter ! In my case I used another app that reads credit cards with NFC and lets you check the logs where you can find these commands.

https://play.google.com/store/apps/details?id=com.github.devnied.emvnfccard

That's the app although there must be a way to get those commands online depending on the card presented, don't know much more about it

tgreco, 2024-02-07:
Anyone interested in where the magic values are:
https://github.com/devnied/EMV-NFC-Paycard-Enrollment/blob/master/library/src/main/java/com/github/devnied/emvnfccard/enums/EmvCardScheme.java#L32

joviksdev, 2024-04-19:
> toHexString

I am getting same issues too, any one with solution to this.

Siti-Shaufi, 2024-05-13:
> @Beejay-bot
> 
> Mastercard example
> 
> ```
> import NfcManager, { NfcTech } from 'react-native-nfc-manager'
> import emv from 'node-emv'
> 
> readMasterCardCreditCard = async () => {
> 	try {
> 		NfcManager.cancelTechnologyRequest()
>     } catch (error) { }
> 	
> 	try {
> 		const commands = [
> 			"00A4040007A00000000410100E",
> 			"80A8000002830000",
> 			"00B2011400",
> 			"00B2010C00",
> 			"00B2012400",
> 			"00B2022400"
> 		]
> 		
> 		await NfcManager.requestTechnology([NfcTech.IsoDep])
> 		
> 		const responses = []
>       
> 		for (let i = 0; i < commands.length; i++) {
> 			const resp = await NfcManager.isoDepHandler.transceive(this.toByteArray(commands[i]))
> 			responses.push(resp)
> 		}
> 
> 		if (responses && responses.length > 3) {
> 			const r = await this.getEmvInfo(this.toHexString(responses[3]))
> 			if (r) {
> 				const cardInfo = this.getCardInfoMasterCard(r)
> 				if (cardInfo) {
> 					return {
> 						card: cardInfo.card,
> 						exp: cardInfo.exp
> 					}
> 				} else {
> 					return null
> 				}
> 			} else {
> 				return null
> 			}
> 		} else {
> 			return null
> 		}
>     } catch (error) {
> 		return null
>     } finally {
> 		NfcManager.cancelTechnologyRequest()
> 	}
> }
> 
> getEmvInfo = (info) => {
>     return new Promise((resolve) => {
> 		emv.describe(info, (data) => {
> 			if (data) {
> 				resolve(data)
> 			} else {
> 				resolve(null)
> 			}
> 		})
>     })
> }
> 
> toByteArray = (text) => {
> 	return text.match(/.{1,2}/g).map(b => {
> 		return parseInt(b, 16)
> 	})
> }
> 
> toHexString = (byteArr) => {
>     return byteArr.reduce((acc, byte) => {
> 		return (
> 			acc + ('00' + byte.toString(16).toUpperCase()).slice(-2)
> 		)
>     }, '')
> }
> 
> getCardInfoMasterCard = (responses) => {
>     let res
>     let end = false
>     for (let i = 0; i < responses.length; i++) {
> 		const r = responses[i]
> 		if (r.tag === "70" && r.value && r.value.length > 0) {
> 			for (let j = 0; j < r.value.length; j++) {
> 				const e = r.value[j]
> 				if (e.tag === "5A" && e.value) {
> 					if (!res) {
> 						res = {
> 							card: e.value
> 						}
> 					} else {
> 						res.card = e.value
> 					}
> 
> 					if (res.card && res.exp) {
> 						end = true
> 					}
> 				}
> 
> 				if (e.tag === "5F24" && e.value) {
> 					if (!res) {
> 						res = {
> 							exp: e.value
> 						}
> 					} else {
> 						res.exp = e.value
> 					}
> 
> 					if (res.card && res.exp) {
> 						end = true
> 					}
> 				}
> 
> 				if (end) {
> 					break
> 				}
> 			}
> 
> 			if (end) {
> 				break
> 			}
> 		}
>     }
>     return res
> }
> ```

Hi, already try this but not working. 
Log shown 
```
lenHex: 82 lenBin: 10000010 ---> len is more than 1 byte
length (decimals): 0 offset: 8 - this is for the substring of emv_data
``` 

Is this because of card or the env.|

Thanks

Siti-Shaufi, 2024-05-14:
Hi @lmatamoros 

Do you have any idea for amex cards? The emv result in empty.
This is the commands that I used.

```
const commands = [
        "00A404000E325041592E5359532E444446303100", // Select PPSE
        "00A4040007A0000000250101", // Select Amex application
        "A000000025", // American Express
        "A0000000250000", // American Express
        "A00000002501", // American Express
        "A000000025010104", // American Express
        "A000000025010402", // American Express
        "A000000025010701", // American Express
        "A000000025010801", // American Express
      ];
``` 
Thanks

altaga, 2024-09-01:
> @Beejay-bot
>
> Mastercard example
>
> ```
> import NfcManager, { NfcTech } from 'react-native-nfc-manager'
> import emv from 'node-emv'
>
> readMasterCardCreditCard = async () => {
> 	try { [...]

Please i really need Mastercard Reading, there is no way to get the Mastercard values? I can read Visa , but no Mastercard.