revtel/react-native-nfc-manager #174: Sending transceive when tag is detected

ChristopheBelpaire, 2019-07-23:
Hello,
I'm trying to send a transceive command when the tag is detected like this  (the goal is to send the apdu when the tag is detected):
```
    NfcManager.registerTagEvent(this._onTagDiscovered)

_onTagDiscovered = tag => {
    console.warn('Tag Discovered', tag);
    console.warn(tag.id);
    NfcManager.requestTechnology(NfcTech.NfcV)
    .then(() =>{
      console.warn("technology discovered")
      NfcManager.getTag()
    })
    .then(tag =>{
      console.warn("Technology requested")
      NfcManager.transceive([0x02,0x35,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00])
     }
    )
    NfcManager.transceive([0x02,0x35,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00])
    .then(rapdu => {
      console.warn(rapdu);
    });

```

But it doesn't seems to work.
Am I doing something wrong ?

bigbellyburger, 2019-08-04:
I'm having a similar issue where `NfcManager.requestTechnology` appears to block within the `registerTagEvent` callback. It seems to block until a new tag is detected, and does not operate on the currently scanned tag.

whitedogg13, 2019-08-22:
Hi @ChristopheBelpaire sorry for the late response.

I think the problem might be that missing promise chaining, so the command doesn't go sequentially
```javascript
_onTagDiscovered = tag => {
    console.warn('Tag Discovered', tag);
    console.warn(tag.id);
    NfcManager.requestTechnology(NfcTech.NfcV)
    .then(() =>{
      console.warn("technology discovered")
      return NfcManager.getTag(); // <-- chain the promise!
    })
    .then(async tag => { // or use the async function
      let resp = await NfcManager.transceive([0x02,0x35,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00]);
      console.log('resp', resp);
     })
```

Thanks!

yanielbf, 2019-09-12:
Hi, I implemented a class to manager  NFC over this library. This is my class:

```
import NfcManager, { NfcTech } from "react-native-nfc-manager";

class NFCManagerImpl {

  hasStartedNFC = false;
  callback = null;

  constructor(){}
  
  init = (callback) => {
    this.callback = callback;
    this.startNFC();
  }

  startNFC = async () => {
    const isSupported = await this._isNFCSupported();  
    if (isSupported && !this.hasStartedNFC) {
      const startResult = await this._startNFCManager();
      if (startResult.status === 'success') {
        this.hasStartedNFC = true;
        this.registerTagEvent();        
        return this.callback(startResult);
      }
      return this.callback(startResult);
    }
    return this.callback({
      status: "error",
      message: "Device no support NFC"
    });
  };

  registerTagEvent = () => {
    // First array applet Id
    // Second array select file
    // Thrid array read binary
    return NfcManager.registerTagEvent(()=>{})
    .then(() => NfcManager.requestTechnology(NfcTech.IsoDep))    
    .then(res => NfcManager.getTag())
    .then(() => NfcManager.transceive([0x00, 0xA4, 0x04, 0x00, 0x0C, 0xA0, 0x00, 0x00, 0x00, 0x18, 0x40, 0x00, 0x00, 0x01, 0x63, 0x42, 0x00, 0x00]))
    .then(() => NfcManager.transceive([0x00, 0xA4, 0x04, 0x00, 0x02, 0x70, 0x01]))
    .then(() => NfcManager.transceive([0x00, 0xB0, 0x00, 0x00, 0xFF]))
    .then(res => console.log('RES', res))
    .catch(err => console.log(err));
  };

  stopNFC = () => {
    if (this.hasStartedNFC) {
      this._unregisterTagEvent();
      this._stopNFCManager();
      this.hasStartedNFC = false;
      this.callback({
        status: 'success',
        message: 'NFC stoped'
      });
    }
  };

  _isNFCSupported = async () => {
    return NfcManager.isSupported();
  }

  _startNFCManager = async () => {
    return NfcManager.start().then(res => ({
      status: 'success',
      message: 'NFC started'
    })).catch(err => ({
      status: 'error',
      message: err
    }));
  }

  _unregisterTagEvent = () => {
    NfcManager.unregisterTagEvent();
  };

  _stopNFCManager = () => {
    NfcManager.stop();
  };

}

export const nfc = new NFCManagerImpl();
```
This is my code in App.js 

```
import React, { useEffect, useState } from 'react';
import { View, Text, Button } from 'react-native';

import { nfc } from './NFCManagerImpl';

const App = () => {

  const [value, setValue] = useState({});
  
  const handleCallBack = (data) => {
    setValue(data);
  }
  
  useEffect(() => {
    nfc.init(handleCallBack);
  }, [])

  return (
    <View style={{alignItems: 'stretch', padding: 20}}>
      <Text>Data scanned: {JSON.stringify(value)}</Text>
      <View style={{flexDirection: 'row', marginTop: 15}}>
        <Button style={{flex: 1}} title="Start NFC" onPress={nfc.startNFC} />
        <Button style={{flex: 1}} title="Stop NFC" onPress={nfc.stopNFC} />
      </View>    
    </View>
  );
};

export default App;
```
The problem is that when i scan a card Id only return a array of two position like  [105, 134]. I don't have idea how interpret this. Help me please

ncuthbert, 2019-09-28:
@yanielbf convert those numbers to hexadecimal (105, 134 = 0x69, 0x86). This is an APDU response code (SW1, SW2) corresponding to **ERROR: Command not allowed (no current EF)**

Here is a list of APDU response codes:
https://www.eftlab.com/knowledge-base/complete-list-of-apdu-responses/

I've just started digging into to this and working on a successful APDU command to retrieve card details. If you crack it, let me know 👍

yanielbf, 2019-09-29:
Hi @ncuthbert, thank you for your answer. I am stopping right now with this solution. I next week I will try again but I have an interest in develop a solution to read identity card Uruguay. I will contact you for solving this trouble. Thank you again for your help!!!!

closed 2021-05-26 by github-actions[bot]