revtel/react-native-nfc-manager #180: I can't stop detection and start it again

reza7rm, 2019-08-16:
Hi, first of all, thanks for this fantastic library. I have an issue that when I want to stop tag detection and then invoke it again.
 When I use unregister tag event and then register again using registerTagEvent I get the warning "You can only issue one request at a time" and the scan doesn't work anymore until I close the app and open it again , when I use 
cancelTechnologyRequest I get Illegal callback invocation from native module error and app crashes. What is the best way to stop Detection and start it again?

whitedogg13, 2019-08-18:
Hi @reza7rm I just found an issue which might cause app to crash when calling `cancelTechnologyRequest`. Please try v1.2.3 and see if it's fixed.

Thanks!

whitedogg13, 2019-08-24:
Just an update, there's some bug in v1.2.3 release, please use v1.2.4 instead.

closed 2021-05-26 by github-actions[bot]