revtel/react-native-nfc-manager #181: error: package android.support.annotation does not exist

dotpegaso, 2019-08-20:
```
Task :react-native-nfc-manager:compileDebugJavaWithJavac FAILED
/node_modules/react-native-nfc-manager/android/src/main/
java/community/revteltech/nfc/NfcManager.java:10: error: package android.support.annotation does not exist
import android.support.annotation.Nullable;
```

Getting this error using React Native 0.59.4

I think it's something related to AndroidX. 
[Jetify](https://github.com/mikehardy/jetifier#usage-for-source-files) didn't solve this tho.

whitedogg13, 2019-08-20:
@dotpegaso sounds like that, can you try to use `npm i --save react-native-nfc-manager@beta` to see if it works? I have include the androidX support in beta release, because I thought it is only required in RN >= 0.60.

dotpegaso, 2019-08-21:
Works, thanks!

closed 2019-08-22 by whitedogg13