revtel/react-native-nfc-manager #185: iOS Beta 2

jafri, 2019-08-25:
I see that you use

```
      let resp = await NfcManager.requestTechnology('ndef');
```

Shouldn't the tech be NfcTech.Ndef which resolves to 'Ndef'

whitedogg13, 2019-08-25:
Good catch! That’s because currently the two(iOS/android) platforms are different in their API design, so we want to limit the impact for Android during iOS 13 dev to avoid any accidents, even for simple defines.

Of cause, we will eventually try to sync up the exposed APIs for both platforms, but maybe in later dev stage.

jafri, 2019-08-25:
I see @whitedogg13 so should I be using `ndef` for iOS and `Ndef` for android?

whitedogg13, 2019-08-25:
Sadly that’s a yes for now. Thanks for the understanding^^

jafri, 2019-08-25:
No problem @whitedogg13, one more question. `closeTechnology`, is that now available on IOS, like `cancelTechnologyRequest`? Or is it not needed on iOS

whitedogg13, 2019-08-25:
Not available, please use `cancelTechnologyRequest` for now.

closed 2019-08-25 by jafri