revtel/react-native-nfc-manager #188: writing data for nfc tag to configure wifi on Tag

deenMuhammad, 2019-08-26:
I am trying to write wifi information in application/vnd.wfa.wsc format.
how can I achieve this?

deenMuhammad, 2019-08-26:
is there any utility to make payload wifi compatible?

deenMuhammad, 2019-08-27:
I seem to have come up with a functionality that can enable the module to write wifi information to let the tag become connectable to wifi when tagged!

deenMuhammad, 2019-08-27:
`@ReactMethod
    public void requestNdefWifiWrite(ReadableArray rnArray, ReadableMap options, Callback callback) {
        // System.out.print(rnArray);
        synchronized(this) {
            if (!isForegroundEnabled) {
                callback.invoke("you should requestTagEvent first");
                return;
            }

            if (hasPendingRequest()) {
                callback.invoke("You can only issue one request at a time");
            } else {
                boolean format = options.getBoolean("format");
                boolean formatReadOnly = options.getBoolean("formatReadOnly");

                //try {
                    NdefMessage msgToWrite;

                    /// the only case we allow ndef message to be null is when formatting, see:
                    /// https://developer.android.com/reference/android/nfc/tech/NdefFormatable.html#format(android.nfc.NdefMessage)
                    ///	this API allows the `firstMessage` to be null
                    if (format && rnArray == null) {
                        msgToWrite = null;
                    } else {
                        // String[] arr = new String(2);
                        // arr[0] = "123";
                        // arr[1] = "123";
                        String[] arr = {rnArray.getString(0),rnArray.getString(1)};
                        byte[] payload = generateNdefPayload(arr);
                        //byte[] bytes = rnArrayToBytes(rnArray);
                        //msgToWrite = new NdefMessage(bytes);
                        NdefRecord mimeRecord = new NdefRecord(
                        NdefRecord.TNF_MIME_MEDIA,
                        NFC_TOKEN_MIME_TYPE.getBytes(Charset.forName("US-ASCII")),
                        new byte[0],
                        payload);
                    
                        msgToWrite = new NdefMessage(new NdefRecord[] {mimeRecord});
                    }

                    writeNdefRequest = new WriteNdefRequest(
                            msgToWrite,
                            callback, // defer the callback
                            format,
                            formatReadOnly
                    );
                //} catch (FormatException e) {
                //    callback.invoke("Incorrect ndef format");
                //}
            }
        }
    }

private static byte[] generateNdefPayload(String[] arr) {
        String ssid = arr[0]; // wifi ssid
        short ssidSize = (short) ssid.getBytes().length;

        short authType;
        authType = AUTH_TYPE_WPA2_PSK;

        String networkKey = arr[1]; // wifi password
        short networkKeySize = (short) networkKey.getBytes().length;

        byte[] macAddress = new byte[MAX_MAC_ADDRESS_SIZE_BYTES];
        for (int i = 0; i < MAX_MAC_ADDRESS_SIZE_BYTES; i++) {
            macAddress[i] = (byte) 0xff;
        }

        /* Fill buffer */

        int bufferSize = 18 + ssidSize + networkKeySize; // size of required credential attributes

        ByteBuffer buffer = ByteBuffer.allocate(bufferSize);
        buffer.putShort(CREDENTIAL_FIELD_ID);
        buffer.putShort((short) (bufferSize - 4));

        buffer.putShort(SSID_FIELD_ID);
        buffer.putShort(ssidSize);
        buffer.put(ssid.getBytes());

        buffer.putShort(AUTH_TYPE_FIELD_ID);
        buffer.putShort((short) 2);
        buffer.putShort(authType);

        buffer.putShort(NETWORK_KEY_FIELD_ID);
        buffer.putShort(networkKeySize);
        buffer.put(networkKey.getBytes());

        return buffer.array();
    }`

deenMuhammad, 2019-08-27:
```java
private static byte[] generateNdefPayload(String[] arr) {
        String ssid = arr[0]; // wifi ssid
        short ssidSize = (short) ssid.getBytes().length;

        short authType;
        authType = AUTH_TYPE_WPA2_PSK;

        String networkKey = arr[1]; // wifi password
        short networkKeySize = (short) networkKey.getBytes().length;

        byte[] macAddress = new byte[MAX_MAC_ADDRESS_SIZE_BYTES];
        for (int i = 0; i < MAX_MAC_ADDRESS_SIZE_BYTES; i++) {
            macAddress[i] = (byte) 0xff;
        }

        /* Fill buffer */

        int bufferSize = 18 + ssidSize + networkKeySize; // size of required credential attributes

        ByteBuffer buffer = ByteBuffer.allocate(bufferSize);
        buffer.putShort(CREDENTIAL_FIELD_ID);
        buffer.putShort((short) (bufferSize - 4));

        buffer.putShort(SSID_FIELD_ID);
        buffer.putShort(ssidSize);
        buffer.put(ssid.getBytes());

        buffer.putShort(AUTH_TYPE_FIELD_ID);
        buffer.putShort((short) 2);
        buffer.putShort(authType);

        buffer.putShort(NETWORK_KEY_FIELD_ID);
        buffer.putShort(networkKeySize);
        buffer.put(networkKey.getBytes());

        return buffer.array();
    }
```

deenMuhammad, 2019-08-27:
```java
//new statics 
    public static final short CREDENTIAL_FIELD_ID = 0x100e;
    public static final short SSID_FIELD_ID = 0x1045;
    public static final short AUTH_TYPE_FIELD_ID = 0x1003;
    public static final short AUTH_TYPE_WPA2_PSK = 0x0020;
    public static final short NETWORK_KEY_FIELD_ID = 0x1027;
    public static final int MAX_SSID_SIZE_BYTES = 32;
    public static final int MAX_MAC_ADDRESS_SIZE_BYTES = 6;
    public static final int MAX_NETWORK_KEY_SIZE_BYTES = 64;
    private static final String NFC_TOKEN_MIME_TYPE = "application/vnd.wfa.wsc";
    //new end
```

deenMuhammad, 2019-08-27:
Whenever you tag nfc your phone prompts message to connect to wifi.
I have tested the new function and it works

whitedogg13, 2019-08-28:
hi @deenMuhammad sorry for the late resonse!

It will be an awesome utility if we have that! However, I think this kind of utility is better to be put in JS side, so both iOS and Android can benefit. If you're available, please feel free to submit that PR, and I will be very welcome to accept that!

deenMuhammad, 2019-08-28:
i will try to do that!
I have one more question!
I am trying to put password authentication to my mifare Ultralight nfc tag, do you think that is possible using existing transceive method?

whitedogg13, 2019-08-28:
Yes, I think `transceive` method is low-level enough to achieve that. I’m

deenMuhammad, 2019-08-29:
> Yes, I think `transceive` method is low-level enough to achieve that. I’m
I have done the password protection using the . transceive . function and it saved my life!
But i had to alter the internal java function to achieve that, at first. but now i do think that it is even possible without altering the existing code. The only thing that i had to do is to understand how the nfc and Mifare Ultralight type of nfc memory structure is constructed!

deenMuhammad, 2019-08-29:
I have used these resources: 
1. https://www.nxp.com/docs/en/data-sheet/NTAG213_215_216.pdf
2. https://stackoverflow.com/questions/22719465/ntag212-mifare-ultralight-with-authentication

closed 2021-05-26 by github-actions[bot]