revtel/react-native-nfc-manager #189: [V2] Use of undeclared identifier 'NFCTagTypeFeliCa'

2joocy, 2019-08-28:
Currently getting our app up to speed on 0.60.5. Android works fine, but IOS seems to be complaining about the following:

```
node_modules/react-native-nfc-manager/ios/NfcManager.m:140:32: error: use of undeclared identifier 'NFCTagTypeFeliCa'
        } else if (tag.type == NFCTagTypeFeliCa) {
                               ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
11 warnings and 20 errors generated.
```

Followed steps to grant nfc permissions etc etc, although this seems like something else?

2joocy, 2019-08-28:
We're currently using 10.3, would we require version 11 for this to work sufficiently?

whitedogg13, 2019-08-28:
@2joocy it seems like your XCode doesn't recognize this type, are you using XCode 11 (beta version)?

whitedogg13, 2019-08-28:
Only if you want to use the upcoming iOS13 NFC features. Otherwise you can use `2.0.0-beta.1`.

roynes, 2019-09-06:
I have this error too while trying the `1.2.4`

whitedogg13, 2019-09-10:
Hi, for current dev, please follow this version guideline:

* XCode 11 beta + RN60: all 2.0.0-beta.X version should be fine
* XCode 10 + RN60: 2.0.0-beta.1
* XCode 10 + RN59 and below: 1.2.2

Thanks!

closed 2021-05-26 by github-actions[bot]