revtel/react-native-nfc-manager #190: Cannot find protocol declaration for 'NFCTagReaderSessionDelegate'

Zamfi99, 2019-08-29:
Hello, I am trying to read some NFC Tags, but it gives me this error when I'm trying to build the project from XCode to my Iphone 8.

XCode version: 10.3
iOS version: 12.4
[image]

whitedogg13, 2019-08-30:
Hi, you can use v1.2.2 for now if you don’t need the upcoming iOS13 features (which requires Xcode 11 beta). Thanks.

Zamfi99, 2019-08-30:
Yep, you're right, I could figure it out how to make it work even with XCode10.3 (with the latest @beta version). Also, on ndef-lib/ndef-text.js you can add the UTF-8 decoder (it tooked me a while until I figured out that is it decoding for UTF-16):
`var out, i, len, c;
        var char2, char3;

        out = "";
        len = data.length;
        i = 0;
        while(i < len) {
            c = data[i++];
            switch(c >> 4)
            {
                case 0: case 1: case 2: case 3: case 4: case 5: case 6: case 7:
                out += String.fromCharCode(c);
                break;
                case 12: case 13:
                char2 = data[i++];
                out += String.fromCharCode(((c & 0x1F) << 6) | (char2 & 0x3F));
                break;
                case 14:
                    char2 = data[i++];
                    char3 = data[i++];
                    out += String.fromCharCode(((c & 0x0F) << 12) |
                        ((char2 & 0x3F) << 6) |
                        ((char3 & 0x3F) << 0));
                    break;
            }
        }

        return out;`

closed 2019-09-10 by whitedogg13