revtel/react-native-nfc-manager #191: Help!!

yanielbf, 2019-08-30:
I copied your example code. When receiving the event only send me an object with this format 
{typeTech: [isoDep, TechA], id: xxxxxxx}. How to parse this result. Help me. Thank

whitedogg13, 2019-09-05:
@yanielbf Sorry for the late response. For `IsoDep` tech type, you will need to send APDU to the tag in order to communicate with it. There’re no standard way to `parse` this kind of tag. You can refer to [here](https://www.google.com.tw/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&ved=2ahUKEwiB0oWlxrnkAhW8yYsBHQo2CM4QFjAAegQIAxAB&url=https%3A%2F%2Fdeveloper.android.com%2Freference%2Fandroid%2Fnfc%2Ftech%2FIsoDep&usg=AOvVaw3T0VH8TMvH8wquLk7WOeHi) for more info.

tcagkansokmen, 2021-02-06:
@whitedogg13 hi bro, how can we get APDU to send as command?

I'm trying to decode message on id card, but it returns non-meaningful data to me.

How can I get name-surname etc?

closed 2021-05-26 by github-actions[bot]

Abramovick, 2025-06-08:
> how can we get APDU to send as command?

Hi @tcagkansokmen, Did you manage to figure this out?