revtel/react-native-nfc-manager #193: Native module cannot be null

roynes, 2019-09-05:
**RN Version**: _0.60.5_
**This package version**: _1.2.4_

I've following manual linking since it won't work with auto linking.

but this problem still came out.

[image]

Is there anyway to fix this?

roynes, 2019-09-05:
Additional errors in compiling.

[image]

[image]

whitedogg13, 2019-09-05:
@roynes For RN60 + XCode 10, please use `2.0.0-beta.1`. Thanks!

roynes, 2019-09-05:
@whitedogg13, tried it to build via xcode and it failed.

whitedogg13, 2019-09-10:
@roynes Sorry for the late response. I have just test in on RN60.5, and it's OK. 

Here's my step for your reference:
```
react-native init NfcTestAppRn60 --version react-native@0.60.5
cd NfcTestAppRn60
npm install --save react-native-nfc-manager@2.0.0-beta.1
cd ios
pod install
```
Then open the `xcworkspace` file (not previous `xcodeproj`) in ios directory and it should work.

whitedogg13, 2019-09-10:
Hi, for current dev, please follow this version guideline:

* XCode 11 beta + RN60: all 2.0.0-beta.X version should be fine
* XCode 10 + RN60: 2.0.0-beta.1
* XCode 10 + RN59 and below: 1.2.2

Thanks!

roynes, 2019-09-10:
> @roynes Sorry for the late response. I have just test in on RN60.5, and it's OK.
>
> Here's my step for your reference:
>
> ```
> react-native init NfcTestAppRn60 --version react-native@0.60.5
> cd [...]

Hi @whitedogg13 , by the word `open xcworkspace` does it pertain to open via `xcode`?

roynes, 2019-09-10:
@whitedogg13,

I got it to work on the simulator now `iPhone X`

but what I got is this. 

`is NFC Supported? false`

[image]

rahamin1, 2020-05-13:
I am having this problem in the following configurations:

- Xcode 11.3 + RN 59.9 + 1.2.2
- Xcode 11.3 + RN 59.9 + 2.0.0-beta.1

Any idea what to do?

alexcastillo, 2021-01-21:
Same here. My configurations are:

 - Xcode 12.1 + RN 0.63.4 + react-native-nfc-manager 2.2.1

alexcastillo, 2021-01-21:
A quick update:

I was able to fix the issue by adding iOS 14.3 [device support](https://github.com/filsv/iPhoneOSDeviceSupport) to my XCode since my iPhone is running 14.3.

avivku90, 2021-01-21:
a quick question, can you write on an empty nfc tag after adding it? 

> A quick update:
> 
> I was able to fix the issue by adding iOS 14.3 [device support](https://github.com/filsv/iPhoneOSDeviceSupport) to my XCode since my iPhone is running 14.3.

alexcastillo, 2021-01-21:
Hi @avivku90,

I haven't tried that, I'm mostly reading from tags. Maybe @whitedogg13 can help?

whitedogg13, 2021-01-21:
@avivku90 it depends. 

If you're talking about writing NDEF, then you will need to first format the tag so you can write NDEF message into it. 

If you're using low-level NFC API, you should still be able to directly write into the tag's memory.

whitedogg13, 2021-01-21:
Guys, I will close it for now, please feel free to reopen or drop a new one if issue persists.

closed 2021-01-21 by whitedogg13

alexcastillo, 2021-01-22:
Thank you for your help, Richie!