revtel/react-native-nfc-manager #195: EMV reader

JamieKieranMartin, 2019-09-07:
Hi I'm trying to implement an EMV reader into my app to accept payments. So far I'm able to read the tag, however having no luck from that. 

Is there any resources out there to help me in reading the data contained?

Best regards,
Jamie

TheAussieStew, 2020-03-24:
@JamieKieranMartin Hi Jamie, did you get anywhere with this? Just wondering since I'm also trying to achieve the same thing.

JamieKieranMartin, 2020-03-27:
@TheAussieStew I had no further luck unfortunately

paranoia5, 2020-06-14:
@TheAussieStew @JamieKieranMartin have you guys found a solution for this issue?

paranoia5, 2020-06-14:
@JamieKieranMartin I couldn't even read the tag on android? How were able to read the tag? I tried but I got no readings at all. :(((((

jameslevine, 2020-06-17:
Anyone had any further luck on this? I'm trying to implement into a payment process too.

jameslevine, 2020-06-22:
@whitedogg13 Is it possible to read EMV data from a card with this library? When I tried, I got tranceive fail console warning.

closed 2021-05-26 by github-actions[bot]

manishbss-sdei, 2021-08-09:
Did any one find the solution for this issue ?

eyodohan, 2022-12-27:
Did any one find the solution for this issue ?