revtel/react-native-nfc-manager #197: iOS Mifare Error after Success

NilsBaumgartner1994, 2019-09-11:
First of all, thank you for your great work.
Well now my error: First it works perfectly. I can read all my data as i wish, how often i like. But when i cancel the detection i wont be ever able to start it again. Even after a restart and reinstall of the App:

I used your example from: https://github.com/whitedogg13/react-native-nfc-manager/blob/master/example/AppV2Mifare.js

The only way to reenable the function is to restart the iphone.
I tried to analyse the error and I'm pretty sure, that the function: " NfcManager.cancelTechnologyRequest" does not work properly.

Here is a small log from my Xcode:

2019-09-11 21:34:56.825599+0200 Swosy[391:28586] [CoreNFC] 00000002 8294f480 -[NFCTagReaderSession beginSession]:318  error:Error Domain=nfcd Code=2 "Busy" UserInfo={NSLocalizedDescription=Busy}, errorCode: 0xcb
2019-09-11 21:34:56.825704+0200 Swosy[391:23899] NFCTag didInvalidateWithError
2019-09-11 21:34:56.884 [warn][tid:com.facebook.react.JavaScript] you need to call registerTagEventEx first
2019-09-11 21:34:56.883965+0200 Swosy[391:29137] you need to call registerTagEventEx first
2019-09-11 21:34:56.887 [warn][tid:com.facebook.react.JavaScript] Clean Up

As you can see the main problem is, that the Reader seems still to be active somehow ?
So that in fact would mean, that the cleanup does not work as expected.

NilsBaumgartner1994, 2019-09-11:
I also can cause follwoing error:

2019-09-11 21:53:08.074302+0200 Swosy[298:8684] NfcManager.requestTechnology
2019-09-11 21:53:08.249227+0200 Swosy[298:5888] NFCTag didBecomeActive
2019-09-11 21:53:10.976822+0200 Swosy[298:5888] NFCTag didDetectTags
2019-09-11 21:53:10.986782+0200 Swosy[298:7368] [CoreNFC] 00000002 82d3b800 -[NFCTagReaderSession _connectTag:error:]:568  Error Domain=NFCError Code=100 "Stack Error" UserInfo={NSLocalizedDescription=Stack Error, NSUnderlyingError=0x280c19800 {Error Domain=nfcd Code=15 "Stack Error" UserInfo={NSLocalizedDescription=Stack Error}}}
2019-09-11 21:53:11.007 [warn][tid:com.facebook.react.JavaScript] Stack Error
2019-09-11 21:53:11.006949+0200 Swosy[298:8684] Stack Error

whitedogg13, 2019-09-17:
@NicholasBertazzon Sorry for the late response.  

Can you tell me your iOS version / XCode version? Mine are iOS public beta 2 + XCode beta 6, and the example works fine.

BTW, app shouldn't be able to affect operating system after closed, is it a seldom issue or a 100% reproducable one?

NilsBaumgartner1994, 2019-09-26:
Funny thing I know it shouldnt affect it after closing the app but it still was there. I anyway modified my code, so I would have to test it again in the new released iOS 13

tassiovale, 2020-02-01:
I'm also getting the "NFCTag didInvalidateWithError" error message. It works fine on Android. All write steps on iOS seems okay.

kaopath, 2020-11-29:
`        (async () => {
          let tag;

          try {
            let resp = await NfcManager.requestTechnology(NfcTech.MifareIOS, {
              alertMessage: this.props.scanMessage,
            });

            tag = await NfcManager.getTag();
          } catch (ex) {
            console.warn('ex', ex);
          } finally {
            this._cleanUp();
          }
          if (tag) {
            this.nfcTagEventHandler(tag);
          }
        })();`

I am getting this error (NFCError:100,nfcd:15) in that function above. In latest version of IOS.

closed 2021-05-26 by github-actions[bot]