revtel/react-native-nfc-manager #198: Problems with "Update ndef-text.js" commit

NicholasBertazzon, 2019-09-11:
Hi,
I'm here to inform that, after the "Update ndef-text.js" commits (07ed8248d6f721c50ec8f45c377d0326718819cd && 5bfa110d8564e75bc52826abc6bec72f442c899e), now the payload decoded with Ndef.text.decodePayload returns also the language characters (it shouldn't, right? if it should, I think it's a breaking change). Also, it removes characters like §, which was perfectly working before.

Any chance to revert them? Or, if you think that they are useful, I can look at them when I will have some spare time (don't have so much anyway)

whitedogg13, 2019-09-17:
@NicholasBertazzon Thanks for reporting, I will revert them for now!

whitedogg13, 2019-09-25:
Hi @NicholasBertazzon the fixed version has been published, thanks for reporting!

closed 2019-09-25 by whitedogg13