revtel/react-native-nfc-manager #199: `writeNdefMessage` is not working on iOS

deenMuhammad, 2019-09-24:
I have been trying to test writeNdefMessage method on iOS
Xcode: 11.0
React Native: 0.60.5
Device: iPhone 7 with iOS 13 installed
[image]
Is there anything that i am missing?

whitedogg13, 2019-09-26:
@deenMuhammad Sorry for the late response. Is the tag you used formatted?

deenMuhammad, 2019-09-26:
should it be formatted or empty ?

whitedogg13, 2019-09-26:
It should work on tags with existing NDEF message, but the tag needs to be formated, at least from the apple doc [link](https://developer.apple.com/documentation/corenfc/nfcndefstatus/nfcndefstatusnotsupported?language=objc). I havn't tried a unformatted tag yet.

deenMuhammad, 2019-09-26:
i have tried many times in different tags. 
I also tried with iPhone8 and the function is writing 4 times out of 10 times in iPhone8+. 
To me it seems my iPhone 7 is not 100% compatible with the latest CoreNFC sw.
Or It seems to me that CoreNFC has some problem!
What do you think about that?

closed 2021-05-26 by github-actions[bot]