revtel/react-native-nfc-manager #205: 2.0.0-beta.4 exposing iOS invalidate() method

joakimgrr, 2019-10-11:
I have an app that reads and writes data consecutively. If the original read goes trough, but the writing throws an error, the iOS modal still shows a checkmark as if the whole task was done successfully.

Reading trough the iOS NFC documentation, i stumbled upon this: https://developer.apple.com/documentation/corenfc/nfcreadersessionprotocol/3229946-invalidate which seems like something that could be used to determine when a native error screen would be shown.

I didn't find a way to handle this currently, do you think it would make sense to expose this native function or is there other ways to achieve this?

Thank you!

whitedogg13, 2019-10-11:
Hi @joakimgrr thanks for this insight. I think exposing the `invalidateSessionWithErrorMessage` Is the right approach. I will try to add this in the next release.

whitedogg13, 2019-10-24:
@joakimgrr just impl `NfcManager.invalidateSessionWithErrorIOS` in the latest `2.0.0-beta.5`, please try it out!

closed 2019-10-24 by whitedogg13