revtel/react-native-nfc-manager #206: tag content prefixed with 'en'

jothikannan, 2019-10-11:
I am using AndroidTechTestNdef.js from examples folder to read and write the tag using the react-native app. When I try to write a tag, the word 'en' prefixing all the time. It seems to be an issue for me. Can anyone help me to fix this?

Please refer attached images 
[image]
[image]

whitedogg13, 2019-10-20:
Hi @jothikannan sorry for the late response, this issue has been fixed in `2.0.0-beta.4`, thanks!

closed 2019-10-20 by whitedogg13

AlonsoBravo, 2020-12-11:
Same problem here, im using version 2.0.0

crobinson42, 2022-10-04:
This is still an issue.