revtel/react-native-nfc-manager #207: Most stable release for Android only & RN 0.60 +

filippoitaliano, 2019-10-16:
What version should I install if I want the most reliable release for Android and I don't want to support iOS for now? I use RN 0.60.5. Do I have to instal the betaish version or someting else? This is totally unclear...

Thanks.

whitedogg13, 2019-10-20:
I think `2.0.0-beta.4` will be fine, since the beta part is actually targeting iOS 13 and shouldn’t affect Android. Thanks!

filippoitaliano, 2019-10-20:
Thank you for the direction! Probably the decision to implement RN 0.60 support in the beta channel related to iOS development wasn't a great choice because it forced updated projects (with full auto-link support) to use the uncomfortable beta version not knowing exactly the potential instabilities (I can't find any **releases changelog**). 

Anyway, thank you for your work!

closed 2019-10-22 by whitedogg13