revtel/react-native-nfc-manager #208: getLaunchTagEvent() always returns null

Atakspor, 2019-10-21:
This has been asked and discussed before but I still can't get it working. 

I have a NFC text with AAR and plain text data. When the app is closed, reading the NFC tag launches my app but getLaunchTagEvent() always returns null. I have no problem reading the tag when app is already opened. 

Added intent filter to my manifest. 
```<intent-filter>
            <action android:name="android.nfc.action.NDEF_DISCOVERED"/>
            <category android:name="android.intent.category.DEFAULT"/>
            <data android:mimeType="text/plain" />
        </intent-filter>```

I use react-navigation.

levepic, 2019-11-08:
I have the same problem in bare app, contains nothing else but nfc read feature. Tried without intent-filter (with AAR app starts up without intent-filter), other intent-filters, tried release version if something fails because of debug version but still always returns null. When app is in background reading gets app in foreground but getLaunchTagEvent doesnt work neither. Does it work for anyone? Would try a working example.

levepic, 2019-11-09:
Found the solution! It is simple: when you write the tag, the text content should be the FIRST in the ndef message which you filter for. If your intent-filter doesnt match the FIRST record it wont get the tag data. Look here for details: https://stackoverflow.com/questions/25504418/get-nfc-tag-with-ndef-android-application-record-aar/25510642

Atakspor, 2019-11-12:
> Found the solution! It is simple: when you write the tag, the text content should be the FIRST in the ndef message which you filter for. If your intent-filter doesnt match the FIRST record it wont [...]

Thanks a lot! I can't belive this is it.

levepic, 2019-11-12:
One more note on this: if you dont add
`android:launchMode="singleTask"`
and your app is already running the AAR might start your application as another task causing strange behavior (in the log you see the tag read but in the application on the screen it wont update however you have set it in state) - thats what i have experienced.

HannesOberreiter, 2020-02-17:
Hi @levepic,

maybe you can help me out. I'm a little bit confused with this topic or maybe my english is not good enough. I simply can't get the getLaunchTagEvent() working, it always returns null. The app is opening with NFC without problems. 

NFC: Forum Type 2, with a simple URL.

My manifest:

```

  <application
      android:name=".MainApplication"
      android:allowBackup="false"
      android:icon="@mipmap/ic_launcher"
      android:roundIcon="@mipmap/ic_launcher_round"
      android:label="@string/app_name"
      android:theme="@style/AppTheme" >

      <!-- Add this SplashActivity -->
      <activity
        android:name=".SplashActivity"
        android:theme="@style/SplashTheme"
        android:label="@string/app_name"
        android:launchMode="singleTask" >
        <intent-filter>
            <action android:name="android.intent.action.MAIN" />
            <category android:name="android.intent.category.LAUNCHER" />
        </intent-filter>
        <intent-filter>
          <action android:name="android.nfc.action.NDEF_DISCOVERED" />
          <category android:name="android.intent.category.DEFAULT" />
          <data android:host="*.btree.at" android:scheme="https" />
        </intent-filter>
        <intent-filter>
          <action android:name="android.nfc.action.TAG_DISCOVERED" />
          <category android:name="android.intent.category.DEFAULT" />
          <data android:host="*.btree.at" android:scheme="https" />
        </intent-filter>
        <!-- 
        <intent-filter>
          <action android:name="android.intent.action.VIEW" />
          <category android:name="android.intent.category.DEFAULT" />
          <category android:name="android.intent.category.BROWSABLE" />
          <data android:scheme="http" android:host="www.btree.at" />
        </intent-filter>
        -->
      </activity>

      <activity
        android:name=".MainActivity"
        android:label="@string/app_name"
        android:windowSoftInputMode="adjustResize"
        android:configChanges="keyboard|keyboardHidden|orientation|screenSize"
        android:exported="true" >
      </activity>

      <activity android:name="com.facebook.react.devsupport.DevSettingsActivity" />
      
  </application>

</manifest>
``` 

Cheers
Hannes

HannesOberreiter, 2020-02-18:
Ok sorry, found the problem. As I use a splash activity for loading the application, the tag is not inside the .MainActivity available. 

Need to figure out a way to solve this, but it is not a problem of this library or wrong manifest.

Cheers
Hannes

whitedogg13, 2020-02-26:
Hi @HannesOberreiter @levepic it was a nice catch! It will be great if you can add a section in README to describe this!

closed 2020-02-26 by whitedogg13