revtel/react-native-nfc-manager #210: Reading NDEF + UID (iOS13) at the same time

richardszalay, 2019-10-24:
We're using your beta channel to read the UID on iOS (via [the mifare example](https://github.com/whitedogg13/react-native-nfc-manager/blob/master/example/AppV2Mifare.js)), but I've noticed that while the scanned tag includes the UID it doesn't include the NDEF records. Is there a way to pull those at the same time?

whitedogg13, 2019-10-24:
@richardszalay from the iOS exposed API, I don't think we can accomplish that. 

More details: iOS uses different kind of reader sessions for different purposes: NFCNDEFReaderSession for NDEF read/write, and NFCTagReaderSession for low-level tag access like Mifare. While using NFCNDEFReaderSession, we can only obtain NFCNDEFTag, which contains no id. 

That's the current situation, if you get any new ideas, please let me know and I can try it out.

whitedogg13, 2019-10-24:
@richardszalay update, just find a way to accomplish that!

Please use the latest `2.0.0-beta.5`, like this:
```javascript
    try {
      let tech = Platform.OS === 'ios' ? NfcTech.MifareIOS : NfcTech.NfcA;
      let resp = await NfcManager.requestTechnology(tech, {
        alertMessage: 'Ready to do some custom Mifare cmd!'
      });
      console.warn(resp);

      // the NFC uid can be found in tag.id
      let tag = await NfcManager.getTag();
      console.warn(tag);

      // then read NDEF message
      let ndef = await NfcManager.getNdefMessage();  
      console.warn(ndef);

      this._cleanUp();
    } catch (ex) {
      console.warn('ex', ex);
      this._cleanUp();
    }
```

richardszalay, 2019-10-24:
@whitedogg13 Amazing, what an incredible turnaround! I'll give that a go later today.

richardszalay, 2019-10-25:
@whitedogg13 I'm going to do some iOS testing this afternoon, but on Android `NfcManager.getNdefMessage` throws an error in beta5 after calling `await NfcManager.requestTechnology(NfcTech.NfcA)`

whitedogg13, 2019-10-28:
Hi @richardszalay thanks for reporting that, this usage is indeed only work for iOS. 

To accomplish your requirement on both platform, please first upgrade to `2.0.0`, and the NdefMessage along with the `id` should be returned in `getTag` call, so you can omit the following `getNdefMessage` for Android.

richardszalay, 2019-11-02:
@whitedogg13 I'm testing on iOS using 2.0.1 but `NfcManager.getTag()` never resolves.

The code is essentially the same as your sample:

```
await NfcManager.start();

await NfcManager.requestTechnology(NfcTech.MifareIOS, {
  "A message"
})

const tag = await NfcManager.getTag();
```

> iPhone 7
> iOS 13.2
> Tags: 2 different NXP tags

Any ideas on how I can debug from here?

whitedogg13, 2019-11-04:
@richardszalay it sounds wired, since the `getTag` method is not modified between `2.0.1` and `2.0.0-beta.4`, and every path for `getTag` method should be resolved.

Just to clarify, does the iOS NFC prompt show? if it does, will it dismiss after scanning tags?

richardszalay, 2019-11-05:
It does show, but the tag is never scanned. I ended up giving up on this direction as it appears as though I've hit the problem with NXP Classic 4k tags as described in this [SO answer](https://stackoverflow.com/questions/57978507/reading-uids-of-nfc-cards-in-ios-13/58482490#58482490).

Thanks for your help!

closed 2019-11-05 by richardszalay